prusa3d/PrusaSlicer #7593: Rotate object 90 degrees from a right click option?

WhiteDragonD, 2021-12-24:
### Version
2.4.0

### Operating system type + version
Windows 10

### Behavior
I spend a lot of time rotating objects 90 degrees because that's how they are exported from the 3D design program.  So I have to click  on the rotate icon, find the box, drag it around and try to line it up to the 90 degree line.  Using the place on face isn't always an option if the bottom isn't flat.

My request is, to have something like a right click menu on the Rotate (R) option that will let you rotate 90 degrees (maybe 180 also?) on one of the three planes so this task can be done with only two clicks.  

Is this a new feature request? Yes

bubnikv, 2021-12-24:
You can use the page up / down keys to rotate by 45 degrees.

WhiteDragonD, 2021-12-24:
It rotates around on the plate but not on the axis where the red box is.  If you have a statue, for example, a lot of programs will export the file with it laying down on it's back so it's a pain to use the mouse all the time to rotate it up to it's feet.

viper0078, 2021-12-24:
You can select an object and enter an angle to rotate it, but is that not enough?

[image]

It would be nice if each of these three text boxes had a button to rotate it 90 degrees.

foreachthing, 2021-12-26:
> It would be nice if each of these three text boxes had a button to rotate it 90 degrees.

I"d prefer a keyboard shortcut. Blender does it neat: r, x, 90 [enter]. Or r, y, 45 [enter]. It's very fast an the gui doesn't get overloaded. 
(PS: I'v added such buttons to the original Slic3r.. )

closed 2026-09-29 by prusaslicer-bot[bot]