prusa3d/PrusaSlicer #7594: Issue exporting .stl with added negative volume

Pelham3d, 2021-12-24:
### Version
2.4.0
### Operating system type + version
Windows 7
### 3D printer brand / version + firmware version (if known)
Not applicable
### Behavior
Added negative volume doesn't save when export plate as .stl is used
I created a decorative wall hanging. I used an existing stl file and used the add part to flatten the back and added negative volume to create a hole to hang the piece. It slices fine and will print fine. I exported the stl for future use and noticed there is no hole in the resulting stl. The added part is saved to the file but not the hole.

#### Project File (.3MF) where problem occurs
[Angel reworked in prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/7772490/Angel.reworked.in.prusa.zip)

dartrax, 2021-12-29:
> I exported the stl for future use
I would suggest using .3mf for this purpose ;-)
Using STL you not only lose the information what volume is positive and what negative, moreover all volumes will be combined into one item in the prusa part list.
I'm not sure if the stl format has the capability to define different separate volumes and which of those volumes is positive or negative. If it has, it should be added indeed...

fiveangle, 2022-01-03:
Came here to report the same.  "**Export plate as STL...**" exports unmodified original with negative volume modifier ignored.

"**Export plate as STL...**" should export an STL that consists of the features prior to being processed into toolpaths.

bubnikv, 2022-01-03:
PrusaSlicer does not perform boolean operations on 3D models. The negative volumes are subtracted slice by slice in 2D, which is much easier, faster and more robust. Thus export into STL does not subtract the negative volumes.

fiveangle, 2022-01-03:
Then a warning should be presented to the user if export to stl will not function correctly.

artickl, 2022-02-09:
+1 

Exporting expectation is to receive STL with what you have on the screen/print...

dartrax, 2022-02-09:
> Especially counting that no other options exist.

What about exporting .3mf? It contains all informations needed and satisfies the use case of the issue report author.

Pelham3d, 2022-02-23:
I thought this feature was a step toward editing stl files similar to Mattter Cntrl.
Having to jump into Tinkerer's Cad or other such programs to make changes to stl files is time consuming.
Being able to make edits and save the file to that Thing place or the Cult one or Pin Shapely would be a nice addition.
Don't get me wrong I still like using PrusaSlicer. I just got overly excited when I saw this new feature.

pedjas, 2022-08-20:
I stumbled on the same issue today. What is the point having option to export to STL if it just exports unmodified object, which is already available and loaded as STL?

neophyl, 2022-08-20:
> I stumbled on the same issue today. What is the point having option to export to STL if it just exports unmodified object, which is already available and loaded as STL?

Because you could have loaded in an amf file or other format for example and want it as an stl.  Although personally I always use a modelling program to do modelling tasks and leave the slicer to do the slicing.

c0depr1sm, 2022-09-03:
Using PrusaSlicer for quick remixes is such an easy workflow now that we can add and remove volumes!
I wanted to export to STL so I can share remixes based on models that I combined using PrusaSlicer.
Reading this thread, I understand I can share a 3MF file to share a remix, that works for me. 

I got a little issue getting away from STL, although 3MF file format is open and can be red using free open source software (PrusaSlicer). While 3MF is available to everyone for free, it is not (yet) readable by some major 3D tools like Meshlab or Blender (AFAIK). But being open source, anyone can add it to their set of features right?

So we should expect more software will start reading the 3MF files format in the future, especially if it becomes common among 3D model files available online.

Understanding the "boolean operation" happens at the slicing step, I would agree to convert that issue to state that:
"the -export to STL- feature is misleading when volume modifiers are in use. A tip could be displayed pointing out the limitation."

Using 2.5.0 beta1 linux x64 GTK3

mmroden, 2022-10-12:
I work on a team of distributed volunteers, and not everyone has a prusa printer or a prusa-slicer based workflow.  Having the ability to export the fully specified object, negative spaces and all, means that I can continue to collaborate with other people on my team.  My alternative is to recreate some models in fusion, which is not cheap.

plundrigan, 2023-02-21:
As a possible future feature, would it not be possible to offer an Group to New Object feature, similar to TinkerCAD (group). That takes all the parts of an object and merges them together into a new object. Which could then be exported?

The premise is you build all the pieces and parts in Prusaslicer possibly designing an entire object. When you feel like you have everything where you want. You just Group to New Object.  All the parts (and Negative Volumes) are preprocessed through the Slicing function and the final Object is saved back to the Objects list as a New Object. I realize I probably over simplified the process. However, maybe taking a tact like this has not been thought of before.

lukasmatena, 2023-02-21:
@plundrigan The most important part is to actually remove the negative volumes from the mesh, which you hurried in the  "preprocessed through the Slicing function" part. Removing the negatives on slices is simple, but stitching the slices together is not.

There are methods to do the subtraction, some are robust, some are fast. Finding one that is both is challenging. It will be adressed at some point.

plundrigan, 2023-02-21:
@lukasmatena I did say I might have over-simplified. :) I just figured the model has to be created somehow during the slicing function. I think the idea of quick prototyping in the slicer would be great. Create a slab throw in a couple of neg volume cylinders for holes and slice and print. Then when finished if the model works saving as stl for sharing with the community.  I know we just got some great features in 2.6 like adding text on objects. So will be patient and see what comes down the pike in future updates.

lukasmatena, 2023-04-02:
Subtraction of negative meshes during export was implemented in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena

plundrigan, 2023-04-03:
That is great news.  Thanks. From: Lukáš Matěna ***@***.***>Date: Sunday, April 2, 2023 at 8:57 AMTo: prusa3d/PrusaSlicer ***@***.***>Cc: AmazinPistolPete ***@***.***>, Mention ***@***.***>Subject: Re: [prusa3d/PrusaSlicer] Issue exporting .stl with added negative volume (Issue #7594)Subtraction of negative meshes during export was implemented in 2.6.0-alpha6. Closing.

binarybum, 2024-09-15:
> Subtraction of negative meshes during export was implemented in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

Was it? I'm running 2.8.0 and my STL exports never include the negative volumes. It's frustrating.

howbotnik, 2024-09-29:
> > Subtraction of negative meshes during export was implemented in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.
> 
> Was it? I'm running 2.8.0 and my STL exports never include the negative volumes. It's frustrating.

Same here. I even downgraded to the 2.6 to see if it worked but I can't get my dowel holes to export with my STL. Tried it many different ways using "fix model" and "simplify model" and still get the "Unable to perform boolean operation" error.

rohitner-google, 2024-10-18:
> Same here. I even downgraded to the 2.6 to see if it worked but I can't get my dowel holes to export with my STL. Tried it many different ways using "fix model" and "simplify model" and still get [...]

I am using version 2.8.1 and also facing this issue.

plundrigan, 2024-10-22:
I was working.  In 2.7 too. I have not tried again in 2.8+ yet. Sent from my Verizon, Samsung Galaxy smartphoneGet Outlook for AndroidFrom: rohitner-google ***@***.***>Sent: Friday, October 18, 2024 5:55:43 AMTo: prusa3d/PrusaSlicer ***@***.***>Cc: AmazinPistolPete ***@***.***>; Mention ***@***.***>Subject: Re: [prusa3d/PrusaSlicer] Issue exporting .stl with added negative volume (Issue #7594)

Subtraction of negative meshes during export was implemented in 2.6.0-alpha6. Closing.

Was it? I'm running 2.8.0 and my STL exports never include the negative volumes. It's frustrating.

Same here. I even downgraded to the 2.6 to see if it worked but I can't get my dowel holes to export with my STL. Tried it many different ways using "fix model" and "simplify model" and still get the "Unable to perform boolean operation" error.

I am using version 2.8.1 and also facing this issue.