prusa3d/PrusaSlicer #7596: Seam position "Rear" does not correctly place seam

MikePfaff, 2021-12-24:
### Version
2.4.0+x64

### Operating system type + version
macOS 10.15.7 (19H1519)

### 3D printer brand / version + firmware version (if known)
Any printer

### Behavior
* _Describe the problem_
  *  Seam position "Rear" does not correctly place seam at the rear

* _Steps needed to reproduce the problem_
  * Load the attached file "rear_seam_problem.stl" (wrapped in a ZIP file for uploading to GitHub) in PrusaSlicer
  * Use the following settings
    * Print settings: 0.20mm QUALITY
      * Change "Seam position" to "Rear"
    * Filament: Prusament PETG
    * Printer: Prusa MK3S+
  * Click on "Slice now"
* _Expected Results_
  * Seam in a straight line fully at the rear of hemisphere (see screenshot)
  * [image]
* _Actual Results_
  * Seam NOT in a straight line the rear of hemisphere (see screenshot)
  * [image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[rear_seam_problem.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/7774368/rear_seam_problem.stl.zip)

MikePfaff, 2021-12-24:
Interestingly the problem is improved when i only slice one of the hemispheres (Right click on model -> "Split to objects", then delete one hemisphere and click "Slice now"). The seam is still not fully aligned in the rear-most position across the layers, though.
[image]

dartrax, 2021-12-24:
If I remember correctly and if implementation hasn‘t changed since v1.4 they take an imaginary point that is in the middle rear (x=bedsizeX/2) outside of the printbed (y=large) and calculate the distance between seam candidates to this point, the shortest one wins.
This point seems to be fixed and not relative to the objects position.
If you have an Object with a slight curve and this is outside of the middle, the seam will orientate towards the middle.

A workaround for you could be to align the objects not side by side, but one behind the other, so that both share the same X coordinate which is half the bedsize.

A fix would be to align the X coordinate of the imaginary far distance rear point with the object center.
Even better would be to be able to make the coordinates customizable, so you have a seam nearest to x/y setting…

The advantage over seam painting would be that this survives a stl reload from disk operation.

n8bot, 2021-12-24:
A nice option is to used the aligned seam setting. It's not at the rear, but it's much more aligned.

MikePfaff, 2021-12-24:
@dartrax Even with the hemisphere fully centered (using the "Arrange" feature), the seam line is still not fully straight: 
[image]

So assuming that the "imaginary middle rear point (y=large and x=bedsizeX/2)" is still the reference point, i think there must be some other code (maybe some strange way of rounding?) that also affects the seam point calculation. Otherwise, at least for the shape of a hemisphere, i'd expect the seam points on the layers to be aligned in a straight line, because on every layer (with the layers being circles) the closest point to the "imaginary middle rear point" should be non-ambiguous.

MikePfaff, 2021-12-24:
@n8bot Thanks for the suggestion, but i already tried that before reporting the issue and in my case that's not the desired solution, because i'd like to have the seam placed at the actual rear pf the object (above the small "cutout" at the bottom) and nicely aligned. I even tried rotating the object so that the "Aligned" seam effectively shows up in the desired place, but when i rotate the object, the "Aligned" seam will just rotate its position along with the object.

I think to fix this would require both:
1. Calculating the imaginary middle rear point for every object separately
2. Some fix so that the "Rear" seam position X stays the same (resulting in a straight line) for each (circle shaped) layer of a hemisphere. This would probably mean one of two things:
    * Either a better (more consistent/predictable) calculation or rounding logic, so that the seam X position for each layer is calculated correctly (remember: each layer of a hemisphere is a perfect circle, so the seam X coordinate should stay the same across all the circle shaped layers that form a hemisphere)
    * Or if a fix of the calculation/rounding logic is not feasible, maybe even add some new kind of "Rear & Aligned" type of seam that for minor seam X position differences between the layers prefers alignment over the flawed calculation/rounding logic.

dartrax, 2021-12-25:
> remember: each layer of a hemisphere is a perfect circle,

Not really. When you export stl, the circle gets approximated by triangles. The slicer never sees a perfect circle and therefore you end up with polygons in each layer. Every point of the polygon is a potential seam candidate. If you have a polygon line parallel to the x axis at the back, the seam may be at the left or right end of this polygon line, because both points may be of equally distance to the imaginary middle rear point. So a rear & aligned type is something to think about.

Did you try SuSi? It has more seam options, you can even define the xy position like so: https://github.com/supermerill/SuperSlicer/discussions/1816
May be with a little fiddling around the seam can be aligned consistent. But I did not test with your 3mf.

closed 2026-09-29 by prusaslicer-bot[bot]