prusa3d/PrusaSlicer #7598: G-Code Viewer not properly showing G2 or G3 commands

guaddy, 2021-12-24:
### Version
_Version of PrusaSlicer used goes here_

PrusaSlicer 2.4.0

### Operating system type + version

Windows 11 Home

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK3S+ Stock

### Behavior

After creating g-code on any model, and running it through [Arc Welder](https://plugins.octoprint.org/plugins/arc_welder/), the processed G-Code does not properly show on G-Code Viewer. The G2 and G3 commands are usually omitted completely or are improperly shown. The g-code does print without errors however and not as shown in the G-Code Viewer.

#### Project File (.3MF) where problem occurs
[g2g3_problems.zip](https://github.com/prusa3d/PrusaSlicer/files/7775018/g2g3_problems.zip)
[image]

n8bot, 2021-12-24:
G2 and G3 are not yet supported in PS or the G-code viewer.

closed 2022-03-25 by guaddy

n8bot, 2022-03-25:
It seems like g2 and g3 have been added to the visualization for PS 2.5. Or, at least, the work is in progress.