prusa3d/PrusaSlicer #7599: Change Color before first M600

Bupher, 2021-12-24:
### Version
2.4

### Operating system type + version
Edition	Windows 10 Home
Version	21H2
Installed on	‎8/‎26/‎2021
OS build	19044.1415
Experience	Windows Feature Experience Pack 120.2212.3920.0

### 3D printer brand / version + firmware version (if known)
Ender 3, stock firmware,

### Behavior
When using an M600 for color change, I want to set the gcode preview colors to match, but I can only change colors after the first M600. Any layers before the first M600 are stuck as yellow. The color change context menu does not appear when I right click the yellow section of the layer slider, but it does when I right click the section after the M600.

_Is this a new feature request?_
Maybe? I would have thought what I expected would be intended behavior. But if it isn't then yes.

#### Project File (.3MF) where problem occurs
3MF 
[No Visual Color Change.zip](https://github.com/prusa3d/PrusaSlicer/files/7775217/No.Visual.Color.Change.zip)

Video of problem
https://www.youtube.com/watch?v=-hn39OvzdN8

neophyl, 2021-12-25:
First layer colour is defined by the filament or extruder settings. As it’s taken from those profiles. If you update those then the initial filament colour changes.

Bupher, 2021-12-25:
Then I guess this is a feature request. I'd like to be able to set the first color the same way I set the 2nd or 3rd when I use m600's.

neophyl, 2021-12-25:
Please see https://github.com/prusa3d/PrusaSlicer/issues/2346

Bupher, 2021-12-25:
Right, I just think the option should exist to temporarily change the first
segment color when using m600's with the same mechanism that is used for
the rest of the layers.

rtyr, 2022-01-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6985. Closing.

closed 2022-01-03 by rtyr