prusa3d/PrusaSlicer #7602: Versione 2.4.0+win64

antares-01, 2021-12-25:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

antares-01, 2021-12-25:
Sembra che vi sia un problema per la stampa con il cambio colore ...cioè, dopo che si cambia il colore , la stampa riprende ma non estrude più il filamento , fatte varie prove , ma non estrude . 

It seems that there is a problem for printing with color change ... that is, after you change the color, the print resumes but no longer extrudes the filament, made various tests, but does not extrude.

rtyr, 2022-01-03:
Please provide your 3MF project file and g-code.

rtyr, 2022-03-10:
No response. Closing.

closed 2022-03-10 by rtyr