prusa3d/PrusaSlicer #7603: "Send System Info" window fails to send system info

JanTuts, 2021-12-25:
### Version
PrusaSlicer-2.4.0+x64-202112211607

### Operating system type + version
MacOS Sierra 10.12.6 (16G2136)
[PrusaSlicer - System Information.txt](https://github.com/prusa3d/PrusaSlicer/files/7775774/PrusaSlicer.-.System.Information.txt)

### 3D printer brand / version + firmware version (if known)
Prusa MINI (firmware 4.3.4)

### Behavior
The new "Send System Info" window in PrusaSlicer 2.4.0 consistently fails to send my system info on my Mac laptop, which I use for 3D modelling and printing. It doesn't appear to be a network issue, because it does work on my desktop PC.
Having my laptop represented in this system survey is important to me, because I don't want to suddenly lose PrusaSlicer support on my laptop because "there is no demand" from that OS/hardware, where in reality myself and perhaps other people simply couldn't report their OS/hardware.
(Also, the error message text is black text on black background for some reason, but I could highlight it to read it.)
[image]

thijstriemstra, 2021-12-25:
Probably related to the ancient *MacOS Sierra 10.12.6* version and it works fine in MacOS latest version?

RAMilewski, 2023-02-01:
Also fails on MacOS 10.14.6 in both Prusa Slicer 2.5.0 and 2.6.0 Alpha 2.

lukasmatena, 2024-02-19:
I guess it can be closed now, the old macOS systems that this issues seems to be specific to are being phased out.

closed 2024-02-19 by lukasmatena