prusa3d/PrusaSlicer #7604: Feature request: filament volumetric speed modifier must accept percentage, filament volumetric speed must be set per extruder

Goury, 2021-12-25:
### Behavior
You can set maximum volumetric speed per print settings and per filament.

This makes some (but not enough) sense and only somewhat works if you only have one machine or all your hot ends are exactly the same.    
If you have more than one hot end or more than one machine, print setting becomes questionable and filament override becomes at best useless but actually destructive.

How it should be done:    
Maximum possible volumetric flow speed is a property of extruder, thus it must be defined in the extruder settings and not anywhere else.    
Print settings and filament settings should allow adjustment but only via percentage modifier.

Please don't use excuses about backwards compatibility.    
This feature, as it is present now, is broken on the very core logical level.    
There must be a way to efficiently own more than one extruder.

Mjankor, 2022-02-22:
Yes please. I've got 3 machines, one with a high flow hotend. Having to duplicate a bunch of profiles just for that one machine is a pain.

Volumetric limit is primarily a function of the hotend, and should probably be under the extruder settings (especially if we end up with toolchangers that can handle big/small nozzles and high/normal flow hotends.

Then include the option to limit the volumetric flow as a percentage in the filament settings.

hvraven, 2025-01-27:
Also would like to support this. With the newly introduced option to mix different nozzle sizes on the Prusa XL the inclusion of the different nozzle sizes (and standard vs. high flow) becomes even more relevant. It is probably not as straight forward due to the already mentioned dependency on the material type, but a value per nozzle combined with a relative adjustment per material would already be a great improvement.

closed 2026-09-02 by Goury