prusa3d/PrusaSlicer #7606: Even-Odd Sling mode at 0% infill

DtcInFLy, 2021-12-25:
### Version
2.4.0

### Operating system type + version
Windows 10 21H1

### 3D printer brand / version + firmware version (if known)
Prusa Mk2s

### Behavior
Try the new Even-Odd slicing mode with STL from 3DLabprint (P51_WINGL_1).
Set 0% Infill, but preview show some Infill (Solid and Bridge). So result not reach the target as well as Cura.

_Is this a new feature request?  Yes

#### Project File (.3MF) where problem occurs
[image]

bubnikv, 2022-01-03:
Do you have "ensure vertical wall thickness" enabled or disabled? Would you please attach your 3MF?

bubnikv, 2022-01-03:
or send it to vojtech.bubnik@prusa3d.cz if the data is private?

DtcInFLy, 2022-01-03:
Hi, same result with or without "Ensure vertical wall thickness" activated.

DtcInFLy, 2022-01-03:
Here the 3mf file
[P51D_wingL_1.zip](https://github.com/prusa3d/PrusaSlicer/files/7802563/P51D_wingL_1.zip)
.

bubnikv, 2022-01-03:
The infill goes away if you set the number of top / bottom layers to zero.

DtcInFLy, 2022-01-03:
Ok, it's a solution. but don't understand this behavior....
Thank's a lot.

Vince

bubnikv, 2022-01-03:
It is a feature that tries to ensure thickness of horizontal walls. It is questionable whether such feature shall be disabled for zero infill, however we plan to implement more fine grained control of this feature.

DtcInFLy, 2022-01-04:
Ok, thanks for your time for answering.
Great Job was made on Prusa Slic3r. 
Good Luck for futurs implementing. ;-)

closed 2026-09-29 by prusaslicer-bot[bot]