prusa3d/PrusaSlicer #7608: Skirt Height setting should be 1 by default

amarilio, 2021-12-26:
### Version
2.4

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
The Skirt's purpose is to prime the nozzle and get rid of stringing prior to printing the main objects. Additional layers on the skirt can hence fail (because the skirt might have excess material), leaving the nozzle dragging the skirt into the print.

In addition, the additional skirt layers lead to more wasted plastic.

_Is this a new feature request?_
Yes

foreachthing, 2021-12-26:
I prime my nozzle with custom gcode. So printing a skirt is no longer necessary. If you need help with that, let me know. 
 Btw: you can edit and save your own profile. 
No need to force defaults one everyone.

bubnikv, 2021-12-26:
I think you are complaining about the skirt height default of Prusa
profiles. We set the skirt height to two layers because the single layer
skirt is difficult to remove from the PEI sheet.

rtyr, 2021-12-26:
One skirt layer is almost impossible to remove from textured sheet.

amarilio, 2021-12-26:
I found it easy to remove the skirt by sliding the printed object over it.
But I can see how users would struggle trying to use their fingernails.

rtyr, 2022-01-03:
I believe it's answered. The main reason for current setting is textured sheet.

Closing.

closed 2022-01-03 by rtyr