prusa3d/PrusaSlicer #7609: Odd support generation

GitPetri, 2021-12-26:
### Version
2.4.0+win64

### Operating system type + version
Windows 10 (build 19044)

### 3D printer brand / version + firmware version (if known)
MK2.5 MMU2s

### Behavior
* Import IKEA power stripe holder.stl
* Set Supports to 'Everywhere'
* Slice
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
  * Intelligent supports
* _Actual Results_
  * Odd supports generated to the 1st layer (only 1st layer)

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
.3mf, .stl and screenshot of generated support attached

[IKEA power stripe holder.zip](https://github.com/prusa3d/PrusaSlicer/files/7776833/IKEA.power.stripe.holder.zip)
[image]
[IKEA power stripe holder 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7776836/IKEA.power.stripe.holder.3mf.zip)

neophyl, 2021-12-26:
That’s the support base layer, it’s generated to provide a base on the plater to give the support something to stick to.  By default it’s set to 3mm. As the support is sticking out slightly on those areas it’s expanding the base out to the value specified.  Set it to 0 and it should go away

GitPetri, 2021-12-27:
> That’s the support base layer, it’s generated to provide a base on the plater to give the support something to stick to. By default it’s set to 3mm. As the support is sticking out slightly on those areas it’s expanding the base out to the value specified. Set it to 0 and it should go away

In theory I get it, but there is nothing sticking to the support base layer. The supports are generated just on the higher layers i.e. the base layer in this case is irrelevant.

GitPetri, 2022-01-17:
As this is just a very minor issue and I wanted to highlight it in case the dev's weren't aware, I'll close the ticket

closed 2022-01-17 by GitPetri