prusa3d/PrusaSlicer #7610: Error Exporting Profiles: "Value out of range: max_print_height"

Agent-Cain, 2021-12-26:
### Version
2.3.3+win64

### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
Self created 3d Printer Profile

### Behavior
I had my Big 3d Printer Profile open (1000mmx1000mmx2000mm), klicked on export bundle and got that error (see screenshot). As I wanted to switch over to 2.4 I'm not sure if this error is already fixed ... 
[image]

I'll might get a work around that, so just see this as a small notification for an not so common error (couldn't find a similar issue).

Thanks in advance.

dartrax, 2021-12-29:
Max print height seems to be 1200mm in PS 2.4.0, when I enter 2000 this occurs:
[image]
When I edit the printers ini file, I get the same message while exporting.
So this seems to be a feature request setting the max print height range higher.

lukasmatena, 2024-02-19:
Duplicate of #6072. Closing.

closed 2024-02-19 by lukasmatena