prusa3d/PrusaSlicer #7613: prusaslicer 2.4 crashes at start (W32 version)

f2knpw, 2021-12-26:
### Version
_Version of PrusaSlicer used goes here_: Version de l’application:	2.4.0.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version Wibdows seven  32bits
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) : Prusa i3 mk2s
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ : crashes at startup (v2.3 works fine)
* _Steps needed to reproduce the problem_ just launch then APPCRASH
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

Gabri988, 2021-12-26:
@f2knpw Which video card do you mount on your pc?

f2knpw, 2021-12-27:
it's an old one... but working fine

f2knpw, 2021-12-27:
It's an ATI Radeon 4800

I saw your ticket a few days ago.
Seems that these ATI Radeon do not work with latest 2.4 prusaslicer...

Gabri988, 2021-12-27:
same problem .... only beta1 working for me

f2knpw, 2021-12-27:
I have just installed the alpha1 which works fine !

I will try the beta1 ! (thanks for the trick)

f2knpw, 2021-12-27:
Yes beta1 works fine as well

bubnikv, 2022-01-03:
Would you please join testing the builds between beta1 and beta2?
https://github.com/prusa3d/PrusaSlicer/issues/7574#issuecomment-1004148995

bubnikv, 2022-01-04:
Would you please try https://github.com/prusa3d/PrusaSlicer/issues/7574#issuecomment-1004701590 ?
Thanks.

f2knpw, 2022-01-04:
Hi,
Thank you for your help.

I tried to install standalone version 2.4 then I replaced the dll into the prisaslicer directory.
But I got an error message saying (in french) that this dll is not designed to run on windows...
I do have a W32 install is it a 32bits dll ?
[image]

bubnikv, 2022-01-05:
Please test

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0+10-win32-g9bbcfb80b-202201050914.zip

You may need to enter

user prusaslicer
password slicer

if a login window opens

Thank you.

f2knpw, 2022-01-05:
you are a BOSS :-)

It works like a charm
Thanks sooo much

bubnikv, 2022-01-06:
thanks for testing

bubnikv, 2022-01-27:
Fixed with prusa3d/wxWidgets@73f029a
The fix will be released with PrusaSlicer 2.4.1-beta1.

closed 2022-01-27 by bubnikv