prusa3d/PrusaSlicer #7616: Please clarify Layer height in Height Range Modifier

laptop-laptop, 2021-12-27:
### Version
2.4

_Use `About->About PrusaSlicer` for release versions_
2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+

### Behavior
* _Describe the problem_

I usually print at 0.15mm layer height but PrusaSlicer seems to set the first layer to 0.2mm by default even the rest are 0.15mm. 

I want to change some settings only on the first layer. When I activated Height Range Modifier, PS filled in 0 as the 'Start at height', 0.2 as the 'Stop at height' and 0.15 mm as the 'Layer height' automatically. Is 'Layer Height' supposed to be 0.15mm or 0.2mm given that the layer height of the fist layer is 0.2mm by default? Of course I also need to change 'Stop at height' from 2 to 0.2mm.

neophyl, 2021-12-27:
I don’t think You can’t change first layer height using a modifier. You have a ‘layer height’ setting that can be set but the setting that determines first layer height is called ‘first layer height’. So when you create a modifier that only effects layer one the modifier for ‘layer height’ will not come into play as it’s doesn’t have a layer to change the settings of.

laptop-laptop, 2021-12-27:
I know the 'First layer height' can be changed but I think in the past somebody mentioned that there is a reason to leave the first layer at 0.2mm unchanged rather than setting it to 0.15mm. What is the reason? 

When modifying the parameters for the first layer using the Height Range Modifier, am I supposed to set the 'Layer Height' to whatever value that matches the 'First layer height' parameter? So in this case it is 0.2mm.

closed 2026-09-29 by prusaslicer-bot[bot]