prusa3d/PrusaSlicer #7617: New ARM64 appimages does not work on Ubuntu Bionic

theofficialgman, 2021-12-27:
### Version
2.4.0 release

### Operating system type + version
L4T Ubuntu Bionic - aarch64
Nvidia jetson nano, jetson xavier, nintendo switch, etc

### Behavior
Appimage appears to be have been built probably on Debian Buster with glibc 2.28. Bionic has glibc 2.27
`prusa-slicer: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by prusa-slicer)`
If the build was made instead of Ubuntu bionic or xenial this would allow for many more systems to run this appimage

bubnikv, 2021-12-27:
We do not support Ubuntu 18 (Bionic) on ARM with our binary builds.

theofficialgman, 2021-12-28:
I am aware that bionic is not supported. This is a request for it to be supported by changing over your buildsystem to use bionic instead of buster (assuming you are using a debian based system to build your appimages).
if there are issues with that let me know what they are

bubnikv, 2021-12-28:
We are building on Raspberry PI. We are happy to scrap some time and resources to do at least as much in regard to ARM 64bit builds. Supporting older libc means exactly what you mentioned: Set up a new build environment.

closed 2026-09-29 by prusaslicer-bot[bot]