prusa3d/PrusaSlicer #7618: Negative Volume Intersecting More Than One Color On Painted Model Crashes to Desktop

MadMax389, 2021-12-27:
### Version
2.4.0 Release

### Operating system type + version
Win 10 x64

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
* A negative volume object intersecting more than one color on a painted .stl will crash to desktop upon slicing with error "PrusaSlicer has encountered a fatal error: Access Violation". If the negative volume object only intersects one color on the .stl, the object slices normally

[image]
[image]
[Negative Volume Crash.zip](https://github.com/prusa3d/PrusaSlicer/files/7780685/Negative.Volume.Crash.zip)
[image]
[image]
[image]

dartrax, 2021-12-29:
I can confirm the crash. Without doing anything it crashes shortly after showing the message, making it hard to take a screenshot...
[image]

bubnikv, 2022-01-04:
@hejllukas 

[image]

hejllukas, 2022-01-05:
Thank you for reporting this issue. It was fixed with 1aac99b809f44dab6abaef2414427e1ce9514c21.

closed 2022-01-05 by hejllukas