prusa3d/PrusaSlicer #7620: Slicing error

Gianpy86, 2021-12-27:
Hello,

I found something that seems a bug during the slicing.

I attached the project (rename it as 3mf).
[prusaslicer-slicingerror.txt](https://github.com/prusa3d/PrusaSlicer/files/7780962/prusaslicer-slicingerror.txt)

[image]

If you tried to slice the model you can see that the slicing process somehow consider some blank part as printable part.

Am I doing something wrong?

thankyou in advance
Regards

Giampiero

----------------

PrusaSlicer version: Version 2.4.0+x64
OS: macOS Monterey 12.0.1 (21A559)

dartrax, 2021-12-29:
I cannot reproduce on Windows 10, may be OS specific?
Can you try repairing the model and slice again (click on the warning Icon next to the model name)

[image]

lukasmatena, 2024-02-19:
Unable to reproduce, possibly outdated. Closing.

closed 2024-02-19 by lukasmatena