prusa3d/PrusaSlicer #7622: Be able to add clearance (offsets) in PrusaSlicer

5jvm0u4, 2021-12-27:
### Version
_Version of PrusaSlicer used goes here_
2.4.0
_Is this a new feature request?_
YES

Just a random thought, wouldn't it be great to be able to add clearance in PrusaSlicer for mating parts instead of in CAD software?

Often the time I found my self designing mating parts, and every time I need to add clearances to the part inside the CAD software. It's not very convenient and reliable considering that offsets required may differ from one printer to another, and also the part size matters too. So it came up to mind that if PrusaSlicer can allow us to add clearance to the mating parts, and the offset value can be calculated inside the slicer or to be user-specified in the printer setting tab, that would be great!

dartrax, 2021-12-27:
If it‘s a parameter that affects the whole model, isn‘t it the same as XY compensation?

What I can imagine is a painting tool to apply an offset to specific (mating) walls.

5jvm0u4, 2021-12-28:
> If it‘s a parameter that affects the whole model, isn‘t it the same as XY compensation?
> 
> What I can imagine is a painting tool to apply an offset to specific (mating) walls.

Yes, that's exactly what I'm thinking of, a painting tool that paints offset, either positive or negative value.

5jvm0u4, 2023-10-20:
For a follow-up, it's there in SuperSlicer as XY compensation and vertical hole shrinking compensation
[image]

closed 2026-09-29 by prusaslicer-bot[bot]