prusa3d/PrusaSlicer #7623: [Enhancement] Add Support material "Style" options to default per-object support material options.

ggppjj, 2021-12-27:
### Version
2.4.0

### Operating system type + version
Windows 11 22523

Hello! I am starting to use the new "snug" supports, and would appreciate if they were available by default when editing a single object's print settings directly, instead of having to add it manually via the Add Settings dialogue.

Thanks for your time!

bubnikv, 2022-01-03:
We are considering making the snug supports a default choice, however we want to collect more data before we make the jump.

ggppjj, 2022-01-03:
Thanks for the reply!

I had seen that making snug default was a consideration, but what I am asking to be considered is for the Style property to be added to the settings that are shown by default when selecting per-item print settings. 

I think screenshots will explain it better:
[image]
[image]
[image]
[image]

When selecting infill settings per-object, Style must be selected explicitly in order to change it.

closed 2026-09-29 by prusaslicer-bot[bot]