prusa3d/PrusaSlicer #7625: Setting Modification Request - "Avoid Crossing Perimeters" path offset inward

3DPmicro, 2021-12-28:
If ACP could track somewhere around the 1st inward perimeter (infill/perimeter interface when using more than one perimeter) that could help significantly with external perimeter quality. I have an older Prusa and the non print moves arent blistering fast so ACP really helps with snotty materials but the current ACP track mainly follows the external perimeter. Any ooze ends up effecting the finish.  Probably not a simple tweak but thanks for your consideration

dartrax, 2021-12-28:
This was mentioned in #5689 but was overshadowed by a bug that‘s fixed now. The feature request is still valid I think.
Fyi, Supermerill implemented that in Susi: https://github.com/supermerill/SuperSlicer/issues/1061

bubnikv, 2022-01-03:
From https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-beta1:

Travel paths planned by the "avoid crossing perimeters" algorithm no more follow the outermost perimeter, but the 2nd outermost perimeter if possible. This should improve the print surface quality and reduce re-heating of overhangs.

closed 2022-01-03 by bubnikv

dartrax, 2022-01-03:
Ah, I did a quick test with 2.4.0 final prior to posting my last post, and I was under the impression that it is not yet implemented in PrusaSlice because I did see travel paths on the outermost perimeter and not the second outermost perimeter. I've added my observation to #5689.