prusa3d/PrusaSlicer #7632: Memory leak when exporting to netfab and cancelling

lablabla, 2021-12-29:
### Version
Version 2.4.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 64Bit

### 3D printer brand / version + firmware version (if known)
Ender 3 V2 (Jyers 2.0.1)

### Behavior
* _Describe the problem_
I have a large model with a notification about 1 open edge. When I right click it and export to netfab for fixing, it took a long time, so I hit the cancel button and the dialog was dismissed. I looked at the memory usage in the Task Manager and the PrusaSlicer started going up and reached over 9GB of memory

* _Steps needed to reproduce the problem_
Load a large mesh with some issue. Right click to export to netfab, after model is exported and started repairing, cancel the job and look at the memory usage

* _Expected Results_
When job is cancelled, no excessive memory usage should be observed

* _Actual Results_
Memory keeps being allocated, reaching several GBs
* _Screenshots from __*PrusaSlicer*__ preview are preferred_
  [https://imgur.com/a/LdjnNQf](url)

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2022-01-03:
Thanks for the report. Skimming through the code I see no error, all the COM objects that are used to interact with the Microsoft API are handled by smart pointers, thus at the time of cancellation (exception is thrown) they shall be destroyed. Maybe I have overlooked something, but maybe the issue is at Microsoft side.

There is one bug I have found: File handle to a file used to write the fixed model to a temp file may not get closed if canceled at the right moment.

lablabla, 2022-01-04:
Thanks,
It happened to me on 2 different machines with 2 different objects. Both STL are ones that I purchased so I can't upload them here, but I will try and generate a simple one and see that it happens with it as well and upload here for anyone else to try and reproduce it.

bubnikv, 2022-01-04:
You may also try the same with Microsoft 3D Builder. We are using the same API as the 3D Builder. If 3D Builder has a memory leak on cancellation, then we cannot do much.

lablabla, 2022-01-04:
I tested 3D Builder and it doesn't have this issue
I captured the screen on both attempts, you can view them here (ignore the memory hogging chrome. it wasn't shut down for like 10 minutes so it's at several GBs.....)
[https://imgur.com/a/mQbk7hu](https://imgur.com/a/mQbk7hu)
I will soon generate a problematic mesh so I can upload here

lablabla, 2022-01-04:
well, looks like a false alarm.
I haven't closed Prusa since capturing the screen and now the memory usage is back to normal, so looks like it only takes pretty long time to cancel and release all the memory, but no apparent memory leak

closed 2022-01-04 by lablabla

bubnikv, 2022-01-04:
> I haven't closed Prusa since capturing the screen and now the memory usage is back to normal, so looks like it only takes pretty long time to cancel and release all the memory, but no apparent memory leak

Releasing memory to the OS is operating system's task.

Thanks for heads up. 
Anyways, I shall fix the issue with possibly leaving the temp file handler open.