prusa3d/PrusaSlicer #7634: Drag stl or 3mf does not work for me anymore with version 2.4

Pierre20230, 2021-12-29:
### Version 2.4.0

### Windows 10 Entreprise 21H1

### M3KS

### Drag stl or 3mf does not work for me anymore with version 2.4
Drag any sttl or 3mf file on the prusa slicer windows and I get the "not possible" icon moiuse (round with a bar)

#### Project File (.3MF) where problem occurs
any file

dartrax, 2021-12-29:
I only get that mouse icon when I drag over the file list in the side bar, the menu bar or the non-modal settings window. All other places (3D window, rest of the sidebar) work. Win10 x64. Can you post a screen shot, may be you have some unusual layout configuration? What's the last version where it worked for you?

Pierre20230, 2021-12-29:
Hi,

 

Thanks for your quick response.

I did not find a way to do a screen shot as I’m holding the mouse while dragging a file (I could send you a video from y phone if you want).

I do have that mouse icon everywhere while dragging including in the 3D windows so impossible to do a drag import (I’m using then the import menu which is working fine but less convenient obviously).
I was working perfectly fine until I upgraded to 2.4.0.

Thanks again for your help (not important but annoying).

Best,

Pierre

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

dartrax, 2021-12-29:
Hi,
no need to take the screenshot while dragging, I just wanted to take a look at the layout config. May be you are using dark mode or tabs or menu items for settings - I don't know if that could have anything to do with this at all, just wanted to make sure.

What version did you use prior to 2.4.0?

Pierre20230, 2021-12-29:
Hello,

 

Here is a screen show, please let me know if that’s what you are looking for (it is French, though, let me know if you want me to switch in English).

I did try various configurations with the same issue.

I’m usually upgrading as soon as possible, I guess my latest version was 2.3.3.

Thanks.

 

Cordialement,

Pierre Cesarini

 ***@***.***> ***@***.***

 

https://www.avast.com/antivirus

dartrax, 2021-12-29:
I think github does not support posting images via mail reply yet.

Pierre20230, 2021-12-29:
[image]

Pierre20230, 2021-12-29:
Sorry, here is it directly on github

dartrax, 2022-01-01:
I was not able to reproduce it, some of the devs need to look into it.
In the meantime, you could simply double click to open a .3mf. Check "Allow just a single PrusaSlicer Instance" in Preferences to avoid opening another window.

closed 2026-09-29 by prusaslicer-bot[bot]