prusa3d/PrusaSlicer #7635: Huge cpu usage when hovering the ruler

Kachidoki2807, 2021-12-29:
**Versions:** PrusaSlicer 2.4.0 64-bit - Win10 Pro 21H2 64-bit

When hovering the ruler, one cpu core is used at 100%.
Just slice a cube, then move the mouse over the ruler, without even dragging it.

I discovered that because the rendering were stuttering while moving around the sliced part with my 3D mouse and simultaneously changing the displayed layer with the 2D mouse to inspect the internals. Then I found that only moving the 2D mouse over the ruler was sufficient to eat up cpu ressources.

dartrax, 2021-12-29:
I can't reproduce, running 2.4.0 64-bit - Win10 Home 21H2 64-bit.
When I use the 3D mouse I shouldn't hover anything outside the preview window because tilting the 3D mouse will send a scroll command to where the 2D mouse currently is, changing dropdown or moving the ruler, but even than no significant CPU usage.
It does not even trigger the GPU there as you can see in the last column in task manager (GPU Module).

[image]

Kachidoki2807, 2021-12-29:
Let me more precise, you must hover the ruler by constantly moving the mouse over it, like when you drag it to inspect layers. When you stop moving, the CPU usage drop down.

When I investigated, I found out that the 3D mouse doesn't have impact, and there is no need to drag the ruler neither.

dartrax, 2021-12-29:
Ok, this is what I get when I constantly move over the ruler:
[image]

This seems like normal behavior to me since it needs to redraw the control constantly.
But yes, I also see the stuttering when using the 3D mouse. However, the stuttering is even worse when hovering the legend or hovering the GCode overlay, while this shows now change in CPU usage. So I think that CPU usage may not be associated with it - look at the GPU usage, here I can see more of a correlation.

Kachidoki2807, 2021-12-29:
Thanks for your feedback.
Indeed, the CPU usage can grow up while hovering some other controls. It might be normal, but it is a pity that moving the complete 3D scene consume less resources than refreshing simple controls.
There is maybe a simple tweak to adjust this, or maybe there is nothing doable to improve this part, and it is not so bothering. That's just for the report.

bubnikv, 2022-01-04:
Updating screen with Win32 API is expensive. We do update the complete scroll bar at each mouse move, which is unnecessary.

[image]

We will try to optimize in the future by redrawing just the region around the mouse cursor.

ristein, 2022-04-01:
I also face this: Only moving the mouse above the ruler without clicking/holding maxes out one cpu-core.
Isn't this caused by recalculating the time to this layer on every hover instead of saving these values?
When hovering somewhere in the preview window without moving anything nor holding/clocking the slicer maxes out my gpu.
I don't understand this at all. There is not even anything clickable here.
My system:
PrusaSlicer Version 2.4.1+win64
Windows 10 21H2
Ryzen 5 2600
GTX 680

murd0cc, 2024-10-16:
Having the same issue!

Moving the cursor over an icon creates a brief (~1 sec) coil whine and barely observable screen/monitor trembling at the same time, then it stops (regardless whether the cursor stays on the icon or leaves). This coincides with GPU use spikes of cca 50%.
This also happens when using the rotate tool and the cursor hovers over the gizmo arrows.

I never noticed this until recently (and I tend to believe it didn't exist until versions 2.8 maybe?!) .

> PrusaSlicer Version 2.8.1
> Windows 11
> Ryzen 9 7950X
> EVGA GTX 970

Kachidoki2807, 2025-04-23:
Bumping up this issue because the default legend (feature type) is maxing out the GPU and 4 CPU threads by doing... nothing. Just let the mouse over a selectable item and don't move.

closed 2026-09-29 by prusaslicer-bot[bot]