prusa3d/PrusaSlicer #7637: Zig Zag line at the boundary of printed part

laptop-laptop, 2021-12-29:
### Version
2.4.0

_Use `About->About PrusaSlicer` for release versions_
2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3MK3S_

### Behavior
The dark green zig zag line indicated by red arrows is imprinted at the boundary in the bottom of the print. As they can't be removed, they created very ugly boundary edge in the first layer. What is such dark green line? Is it possible to disable such line? Without it, what are some possible negative effects on the prints?

https://imgur.com/MgWDLjy

_Is this a new feature request?_

Yes if the ability to disable it is unavailable

neophyl, 2021-12-29:
Try looking at the colour key in the slice preview. It tells you that it’s support interface. So you must be supporting something very close to the bed. 

Disable supports in that area using a blocker if you don’t want support there although I would look at your model geometry and ask yourself if redesigning that area is possible to print without supports is a better use of your time.

bubnikv, 2022-01-04:
Sorry the image is no more and I dont understand the question.

rtyr, 2022-03-10:
No response. Closing.

closed 2022-03-10 by rtyr