prusa3d/PrusaSlicer #7638: Can't add bed termperature to output file name format. "Referencing a vector variable when scalar is expected"

scyto, 2021-12-29:
### Version
2.4

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 11

### 3D printer brand / version + firmware version (if known)
Snapmaker A350

### Behavior
Would like to add major temperatures to file name, i looked at https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)  I added extruder temps just fine.

{bed_temperature}
{first_layer_bed_temperature}

Both generate the following error

```
Failed processing of the output_filename_format template.
Parsing error at line 1: Referencing a vector variable when scalar is expected
{input_filename_base}_{layer_height}mm_{filament_type[0]}_{first_layer_temperature[0]}_{temperature[0]}_{bed_temperature}.gcode
                                                                                                         ^
```

rtyr, 2021-12-29:
This should work `{bed_temperature[0]}`

rtyr, 2022-01-03:
Answered. Closing.

closed 2022-01-03 by rtyr

alber70g, 2024-10-21:
Where are the docs for this?