prusa3d/PrusaSlicer #7639: [Feature request] Filament comparison overview between own with prusa profiles

Devilscave, 2021-12-29:
Hello, I do not know if it's right here, but I do not know myself with the Pull Requests.

I currently use version 2.4.0 and after installing, new configurations have been added once again.
A big problem as I think,
If you change something to the filament settings something you have to always save them in a separate filament profile, if you have already 20 in it, it is not so easy to see all the settings and deviations, to the prusa profiles, to overlook.
There would be a kind of comparison overview of the individual values well, which simply says in which points my profile is different from the Prusa profile. (I hope that was understandable, I work with translator)

dartrax, 2021-12-29:
Something like this?
[image]

Devilscave, 2021-12-29:
WTF, yes that's what I mean! But that's new in the version or? I have probably overlooked.

closed 2021-12-29 by Devilscave

dartrax, 2021-12-29:
Yes, introduced in 2.4.0-alpha1