prusa3d/PrusaSlicer #7640: Feature request

hugomsmarreiros, 2021-12-29:
is it possible to add temperature variation on multiple extruders when using wipe tower?

Thanks in advance.

rtyr, 2022-01-03:
Can you please elaborate? I am not sure I understand your request.

hugomsmarreiros, 2022-01-03:
The option we have when using ooze shield, be able to reduce temperature on the unused extruder, I would like to have that option on wipe tower.

BTW, when using ooze shield, having an option to make a brim for ooze shield itself would be nice, cause if it's a tall part, most of the times the ooze shield becomes losses because of the noozles hitting it with oozing filament.

hugomsmarreiros, 2022-01-03:
To be more clear, my printer has 2 extruders on the same block, so my 2 noozles are constantly going over the print, having the ability to reduce temperature when using wipe tower, would prevent to have mixture of two filaments caused by second extruder even with a big retraction being leaking over the print.

bombela, 2022-12-18:
You could control this with custom gcode. You have all the information available as placeholder variables (previous extruder ID, next one, filament temperature etc).

closed 2026-09-29 by prusaslicer-bot[bot]