prusa3d/PrusaSlicer #7641: Prusa Slicer 2.4 on Apple M1 Max does not generate the right extrusion code

radler64, 2021-12-30:
### Version
2.4.0+arm64

Build: PrusaSlicer-2.4.0+arm64-202112211601
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 12.1 (Build 21C52)
Total RAM size [MB]: 34,360MB

### 3D printer brand / version + firmware version (if known)
Anycubic Mega X

### Behavior
I just changed my Laptop from a Windows 11 Intel based machine to a Mac M1 Max with the operating system above.
I exported the Prusa 2.3.3 settings from the Windows installation using the export command.
Ini-files got created.
I imported those ini files to the  Prusa 2.4.0 installation.
All seemed fine, I can see all my settings.

Than I created a new part with Solidworks, exported it in a 3MF format and imported it to the Slicer.
The part got sliced, all looks fine so far.
I exported it via Octoprint (current release 1.7) to the printer.
The Anycubic started to heat and print.

But there is not enough extrusion!! 
The extrusion is only some micro-millimeters per second!
So after a drive where at least should be 10cm extruded, I would say it extruded may be 10 mm. This leads to nothing.
It looks like, that the code generation does not do the extrusion correctly.
I don't know hot to find out what the problem ist.

I started to print an old model, which was already exported to Octoprint using the Prusa-Slicer before.
Those modells are working.

So what can I do?

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[Prusa-A00710000015-Gehaeuse.zip](https://github.com/prusa3d/PrusaSlicer/files/7791133/Prusa-A00710000015-Gehaeuse.zip)

dartrax, 2021-12-30:
GCode looks ok in preview.
Do you have the possibility to slice again with 2.3.3 and, if this works, post the gcode here for comparison?

radler64, 2021-12-31:
Hello and thanks for your response.

here is the same file, created on my old Win 11 PC with version 2.3.3 with the original settings I had there.
Layer height, start sequence, fan speed and settings and position on the plate might be different.
Thanks for your support
[Prusa-Win11-A00710000015-Gehaeuse_PET_0.6.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/7796194/Prusa-Win11-A00710000015-Gehaeuse_PET_0.6.gcode.zip)
.

dartrax, 2022-01-01:
I started to print the "Prusa-A00710000015-Gehaeuse_PET_0.6.gcode" that you provided in the zip file of the first post, and the Skirt/Brim of the first layer startet to print fine, so I did not print further.
May be there was a mechanical problem with the printer? Or is the underextrusion starting at a higher layer only?

radler64, 2022-01-02:
Thank you so much for your test. 
One question, did you use a printer with a 0.4mm nozzle or - and this is my file for - a 0.6mm nozzle. Could it be, that the nozzle size is not incorporated in the code generation - or is that fact unimportant?
I will check the hardware again...
Thanks Dieter

dartrax, 2022-01-02:
I used a 0.4mm nozzle, but you are using a first layer extrusion width of only 0.3mm, so for this reason I had no problems with a 0.4mm nozzle in the first layer. I think 0.3mm width could be a bit little for 0.6 nozzle, but it is nearly identical in both profiles/gcode files, so if it works with the old gcode file it should also work in the new gcode.

Both gcode files have a nozzle size of 0.6mm in the profile.

It occured to me that the calculated printing times are different:
Old profile: 6h 4m
New profile: 2h 28m
Filament used: ca. 8100mm (both)

So the new profile tries to print the same amount of filament in less than half the time. Could it be that you print too fast and the extruder is struggling to keep up with the speed? The stepper might skip steps or the filament slips. Try to reduce the speed of the new profile and carefully watch the extruder while testing again.

I also print with PETG and I know that you need a powerful extruder if you want to print faster, because of the viscosity of PETG. I use Bondtech Extruder that has a gear ratio to not skip steps and that grips the filament with two gears on both sides to not let it slip.

bubnikv, 2022-06-29:
@dartrax provided analysis, no response from @radler64. Closing.

closed 2022-06-29 by bubnikv