prusa3d/PrusaSlicer #7642: Prusaslicer refuses to close like it used to and will always ask, without the ability to turn it off.

thanhtacles, 2021-12-30:
### Version
_2.4.0_

_Version 2.4.0+win64_

### Behavior
* _Prusaslicer refuses to close and will always prompt you to Save the changes to "xyz" before closing_

    Just drag and drop any 3mf or stl into prusaslicer and try to close it

* _Expected Results_
    Normally in the past, it would just be able to close unconditionally without question. Now it's asking every single time and that's super annoying. This HAS to be a bug or unintended user experience. Dialogue box popup example when xing out of PS:
[image]

And yes, even with all the prompts to ask checked off, it'll still ask you: 
[image]

redtif, 2022-01-03:
I can confirm, same here!

bubnikv, 2022-01-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7114#issuecomment-991132153

closed 2022-01-03 by bubnikv