prusa3d/PrusaSlicer #7644: [Feature request] Ruler and Calibrate

giudiceanto, 2021-12-30:
First time here. I apologise immediately if this is not the correct way to ask for a function.

The following are two tools that would add better manipulation of objects before the slicing.

1)The first is a simple ruler, used to measure the distance between two points within the model (at the moment we can only know the perimeter dimensions of the model) .
I often use it with Microsoft's "3D Builder" software, called Measure tool.

2)The second is a tool for resizing an object on the basis of the distance between two points. The overall scale ratio of the object remains the same. You can find a similar function on Fusion360, when you add a cavas (in 2D unfortunately) and want to calibrate it to have the desired/correct dimensions (scale ratio unaltered).

Example: I have a model of a box that measures 5x3 cm on the outside. On the inside, however, there is an empty space measuring 2x2 cm. In this example case, I would like to resize the object so that the internal empty space measures i.e. 4.2x4.2cm. The box will then be resized so that the internal space is the value I choose, and the external volume accordingly.

thank you

dartrax, 2021-12-30:
Duplicate of #3651

giudiceanto, 2021-12-30:
> Duplicate of #3651

nice! at least I'm not alone! Thank you!

lukasmatena, 2022-01-17:
Closing as a duplicate.

closed 2022-01-17 by lukasmatena