prusa3d/PrusaSlicer #7645: MacOS crash after wakeup from hibernation

Jeggo65, 2021-12-30:
### Version
2.4.0+x64-202112211607

### Operating system type + version
macOS 12.0.1 (Build 21A559)

### 3D printer brand / version + firmware version (if known)
MK3S latest firmware

### Behavior
Crashed after each hibernation of macOS. Happens every time. Latest 2.3.x was OK.
Behaviour is there since first alpha version of 2.4.x 

#### Project File (.3MF) where problem occurs
Not relevant. Happens also without any project.

bubnikv, 2022-01-03:
Anyone else experiences the crash on OSX on dehibernation?

dennyhollick, 2022-01-17:
I am experiencing this exact behaviour. Repeatable 100% of the time.

```
Process:               PrusaSlicer [44738]
Path:                  /private/var/folders/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.4.0+x64-202112211607 (PrusaSlicer-2.4.0+x64-202112211607)
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [44738]

Date/Time:             2022-01-17 00:43:29.582 -0800
OS Version:            macOS 11.6 (20G165)
Report Version:        12

Sleep/Wake UUID:       2DAA548C-9B57-4373-A888-BE25B7E650CF

Time Awake Since Boot: 120000 seconds
Time Since Wake:       49 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x0000000197720e68 __pthread_kill + 8
1   libsystem_pthread.dylib       	0x000000019775343c pthread_kill + 292
2   libsystem_c.dylib             	0x000000019769b454 abort + 124
3   com.prusa3d.slic3r/           	0x000000010204177c wxAbort() + 12
4   com.prusa3d.slic3r/           	0x00000001020d6a30 wxEvtHandler::WXConsumeException() + 192
5   com.prusa3d.slic3r/           	0x00000001020d6944 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 44
6   com.prusa3d.slic3r/           	0x000000010201531c -[wxNonOwnedWindowController windowDidChangeBackingProperties:] + 400
7   com.apple.CoreFoundation      	0x0000000197845f08 __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 28
8   com.apple.CoreFoundation      	0x00000001978e7e5c ___CFXRegistrationPost_block_invoke + 52
9   com.apple.CoreFoundation      	0x00000001978e7dc8 _CFXRegistrationPost + 456
10  com.apple.CoreFoundation      	0x00000001978149d8 _CFXNotificationPost + 720
11  com.apple.Foundation          	0x00000001985a03c0 -[NSNotificationCenter postNotificationName:object:userInfo:] + 64
12  com.apple.AppKit              	0x000000019a8e8b90 -[NSWindow _postWindowDidChangeBackingPropertiesAndDisplayWindowForPreviousBackingScaleFactor:previousColorSpace:] + 212
13  com.apple.AppKit              	0x000000019a283208 __67-[NSWindow _updateSettingsSendingScreenChangeNotificationToScreen:]_block_invoke + 572
14  com.apple.AppKit              	0x000000019a094cf4 NSPerformVisuallyAtomicChange + 140
15  com.apple.AppKit              	0x000000019a282fc0 -[NSWindow _updateSettingsSendingScreenChangeNotificationToScreen:] + 96
16  com.apple.AppKit              	0x000000019a0966cc -[NSWindow _setFrameCommon:display:fromServer:] + 2144
17  com.apple.AppKit              	0x000000019a3e6694 -[NSWindow _adjustWindowToScreen] + 1188
18  com.apple.AppKit              	0x000000019a40da1c -[NSWindow _displayChangedSoAdjustWindows:] + 116
19  com.apple.AppKit              	0x000000019ab6f950 ___NSApplicationPerformScreenInvalidationReactions_block_invoke + 108
20  com.apple.AppKit              	0x000000019a03b918 -[NSApplication enumerateWindowsWithOptions:usingBlock:] + 292
21  com.apple.AppKit              	0x000000019ab6f3cc ___NSApplicationReactToScreenInvalidation_block_invoke + 640
22  com.apple.AppKit              	0x000000019ab6eed0 _NSApplicationReactToScreenInvalidation + 300
23  com.apple.AppKit              	0x000000019ab6f0c4 -[NSApplication(ScreenHandling) _reactToDisplayChanges:] + 396
24  com.apple.AppKit              	0x000000019a72e1ec _NSCGSDisplayConfigurationUpdateAndInvokeObservers + 456
25  com.apple.AppKit              	0x000000019a72e590 _NSCGSDisplayConfigurationDidReconfigureNotificationHandler + 328
26  com.apple.SkyLight            	0x000000019c0288cc (anonymous namespace)::notify_datagram_handler(unsigned int, CGSDatagramType, void*, unsigned long, void*) + 916
27  com.apple.SkyLight            	0x000000019c2ce1d4 CGSDatagramReadStream::dispatchMainQueueDatagrams() + 228
28  com.apple.SkyLight            	0x000000019c2ce0d0 invocation function for block in CGSDatagramReadStream::mainQueueWakeup() + 28
29  libdispatch.dylib             	0x00000001975a2128 _dispatch_call_block_and_release + 32
30  libdispatch.dylib             	0x00000001975a3ec0 _dispatch_client_callout + 20
31  libdispatch.dylib             	0x00000001975b2314 _dispatch_main_queue_callback_4CF + 884
32  com.apple.CoreFoundation      	0x00000001978918e8 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
33  com.apple.CoreFoundation      	0x00000001978502fc __CFRunLoopRun + 2524
34  com.apple.CoreFoundation      	0x000000019784f1a8 CFRunLoopRunSpecific + 600
35  com.apple.HIToolbox           	0x000000019f774b40 RunCurrentEventLoopInMode + 292
36  com.apple.HIToolbox           	0x000000019f7748b4 ReceiveNextEventCommon + 552
37  com.apple.HIToolbox           	0x000000019f774674 _BlockUntilNextEventMatchingListInModeWithFilter + 72
38  com.apple.AppKit              	0x000000019a041660 _DPSNextEvent + 836
39  com.apple.AppKit              	0x000000019a040000 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1292
40  com.apple.AppKit              	0x000000019a031e4c -[NSApplication run] + 596
[... cut ...]
44  com.prusa3d.slic3r/           	0x00000001020402c4 wxAppConsoleBase::MainLoop() + 120
45  com.prusa3d.slic3r/           	0x0000000101fd4a3c wxApp::OnRun() + 36
46  com.prusa3d.slic3r/           	0x000000010208e44c wxEntry(int&, wchar_t**) + 148
47  com.prusa3d.slic3r/           	0x00000001014c73b4 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 292
48  com.prusa3d.slic3r/           	0x0000000100dc9adc Slic3r::CLI::run(int, char**) + 16348
49  com.prusa3d.slic3r/           	0x0000000100dce270 main + 204
50  libdyld.dylib                 	0x0000000197771430 start + 4
```

bubnikv, 2022-01-17:
@dennyhollick 
Was 2.3.3 OK for you as well?

kocikdav, 2022-01-17:
Hello @dennyhollick @Jeggo65. Do you have external monitor connected? Unfortunately we are not able to reproduce this crash. Is there some special Energy Saver settings in System Preferences or any other non-default thing?
From crash dump I'm guessing that WindowDidChangeBackingProperties could mean that Slicer window changed position from non-retina to retina display or otherwise during (de)hibernation.

Jeggo65, 2022-01-17:
As it is a Mac Mini I use an external monitor. I did not made any changes to the energy saving settings.
At the moment the problem seems to be gone. I am not aware of installing any kind of Apple updates.

dennyhollick, 2022-01-18:
@bubnikv The last version worked flawlessly for me. 

@kocikdav It seems to happen regardless of a screen or not for me, But I do often run an external monitor. 

What I did notice -- I get another (potentially unrelated?) crash errors if I run the config wizard also.

```
Process:               PrusaSlicer [24407]
Path:                  /private/var/folders/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.4.0+x64-202112211607 (PrusaSlicer-2.4.0+x64-202112211607)
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [24407]
User ID:               501

Date/Time:             2022-01-17 16:02:04.086 -0800
OS Version:            macOS 11.6 (20G165)
Report Version:        12
Sleep/Wake UUID:       010CD9B3-5D1E-4948-BE7B-C9754A9FD4A6

Time Awake Since Boot: 170000 seconds
Time Since Wake:       1000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000037
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [24407]

VM Regions Near 0x37:
--> 
    __TEXT                      1006bc000-101e08000    [ 23.3M] r-x/r-x SM=COW  /var/folders/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.prusa3d.slic3r/           	0x0000000100fb5360 bool std::__1::operator!=<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > >(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&, std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&) + 48
1   com.prusa3d.slic3r/           	0x0000000100fb47b4 Slic3r::GUI::ConfigWizard::priv::apply_config(Slic3r::AppConfig*, Slic3r::PresetBundle*, Slic3r::PresetUpdater const*, bool&) + 6008
2   com.prusa3d.slic3r/           	0x0000000100fb47b4 Slic3r::GUI::ConfigWizard::priv::apply_config(Slic3r::AppConfig*, Slic3r::PresetBundle*, Slic3r::PresetUpdater const*, bool&) + 6008
3   com.prusa3d.slic3r/           	0x0000000100fb7538 Slic3r::GUI::ConfigWizard::run(Slic3r::GUI::ConfigWizard::RunReason, Slic3r::GUI::ConfigWizard::StartPage) + 448
4   com.prusa3d.slic3r/           	0x0000000100de4544 Slic3r::GUI::GUI_App::run_wizard(Slic3r::GUI::ConfigWizard::RunReason, Slic3r::GUI::ConfigWizard::StartPage) + 148
5   com.prusa3d.slic3r/           	0x00000001019d67d0 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 228
6   com.prusa3d.slic3r/           	0x00000001019d64cc wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
7   com.prusa3d.slic3r/           	0x00000001019d632c wxEvtHandler::ProcessEvent(wxEvent&) + 56
8   com.prusa3d.slic3r/           	0x00000001019d6930 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
9   com.prusa3d.slic3r/           	0x000000010180abb0 wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) + 140
10  com.prusa3d.slic3r/           	0x000000010180aa98 wxMenuBase::SendEvent(int, int) + 152
11  com.prusa3d.slic3r/           	0x00000001018c221c wxMenu::HandleCommandProcess(wxMenuItem*, wxWindow*) + 308
12  com.apple.AppKit              	0x000000019a25a05c -[NSApplication(NSResponder) sendAction:to:from:] + 456
13  com.apple.AppKit              	0x000000019a359360 -[NSMenuItem _corePerformAction] + 444
14  com.apple.AppKit              	0x000000019a359050 -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:] + 100
15  com.apple.AppKit              	0x000000019a3a1484 -[NSMenu performActionForItemAtIndex:] + 196
16  com.apple.AppKit              	0x000000019a3a13a8 -[NSMenu _internalPerformActionForItemAtIndex:] + 100
17  com.apple.AppKit              	0x000000019a3a11b4 -[NSCarbonMenuImpl _carbonCommandProcessEvent:handlerCallRef:] + 116
18  com.apple.AppKit              	0x000000019a33becc NSSLMMenuEventHandler + 728
[... cut ...]
35  com.prusa3d.slic3r/           	0x00000001019402c4 wxAppConsoleBase::MainLoop() + 120
36  com.prusa3d.slic3r/           	0x00000001018d4a3c wxApp::OnRun() + 36
37  com.prusa3d.slic3r/           	0x000000010198e44c wxEntry(int&, wchar_t**) + 148
38  com.prusa3d.slic3r/           	0x0000000100dc73b4 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 292
39  com.prusa3d.slic3r/           	0x00000001006c9adc Slic3r::CLI::run(int, char**) + 16348
40  com.prusa3d.slic3r/           	0x00000001006ce270 main + 204
41  libdyld.dylib                 	0x0000000197771430 start + 4
```

dennyhollick, 2022-01-18:
Just as an update - I deleted all of my application data and started fresh with. Removing my old configuration has seemed to resolve this. I don't think it ported to the new version well.

Specifically, it seems that when I upgraded to this new version, some old configuration from either my 'printer' or 'physical printer' folder was causing crashes both when I awoke from hibernation and when I ran the config assistant. 

I haven't thoroughly tested this yet, but so far so good.

Jeggo65, 2022-01-18:
When I had the crashes, I only upgraded from V2.3 without any changes. Recently I ran the configuration tool to add some filament profiles.
Maybe this „fixed“ the crash problem?

closed 2026-09-29 by prusaslicer-bot[bot]