prusa3d/PrusaSlicer #7647: Export as STL grayed out

kats-greg, 2021-12-30:
### Version
Version 2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
After loading a MMU model, I have one stl that has errors, so I "Fix through the Netfabb", and it successfully fixes the file, and I used to be able to Export as STL to save the changes to the file, but now the Export as STL option is grayed out.  

[image]

bubnikv, 2022-01-04:
@enricoturri1966 has it on his plate.

bubnikv, 2022-01-27:
This will be fixed in PrusaSlicer 2.4.1-beta1. Closing.

closed 2022-01-27 by bubnikv