prusa3d/PrusaSlicer #7654: PrusaSlicer won't change the bed size while changing printer profile.

AssafKatan, 2021-12-31:
### Version
2.4.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 11 #s
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* I have 2 printers, An biqu b1 that I made profile for her and cr10s pro. They have different bed sizes. 235 and 310. On older versions of prusaslicer eveything worked fine, when I switched the printer profile the bed size and height changed like it suppose to  do, but on this new version it is only changing the Z height and I need to change the bed size accordingly.  And yes the printer profile does have the correct sizes.
* change printer profiles with different bed sizes.
  * _If this is a command-line slicing issue, include the options used_
* The bed size to change.
* The bed size wont change
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

dartrax, 2022-01-01:
Works for me, changes the bed size as soon as I switch printers. Could you export your config bundle (Export -> Export Config bundle...) and upload it here? May be there is something special with the printer config.

AssafKatan, 2022-01-01:
> Works for me, changes the bed size as soon as I switch printers. Could you export your config bundle (Export -> Export Config bundle...) and upload it here? May be there is something special with the printer config.

Looks like it is fixed after a restart. Weird but ok, Anyway thanks

closed 2022-01-01 by AssafKatan