prusa3d/PrusaSlicer #7655: Crash on 2.4 on Mac M1 when trying to add new printer

sungo, 2021-12-31:
### Version

2.4.0+arm64

### Operating system type + version

Macbook Air M1 2020, running macOS 12.1

### 3D printer brand / version + firmware version (if known) 

Trying to add a new custom printer

### Behavior

I am attempting to add a new printer / preset in 2.4 and experiencing crashes

1. I entered the configuration assistant via the dropdown in the Printer list in the main view, selected "Add/Remove Printers", then "Add/Remove Presets"

1. Clicked down to Custom Printer. Ticked "Define a custom printer". Entered a profile name. Hit next. 
2. Select "RepRapFirmware" in the firmware type dialog. Hit next.
3. Change shape to "Circular" and specify a diameter. Hit next.
4. Change the nozzle diameter. Next.
5. Change the nozzle and bed temps. Hit finish. 

Always crashes here.

Apple crash log attached.

[crash.txt](https://github.com/prusa3d/PrusaSlicer/files/7795104/crash.txt)

ziwcam, 2022-01-02:
Can generalize this crash a bit further, as I experience a similar issue on my Intel iMac (Retina 5K, 27-inch, 2020). (Stack trace nearly identical)

Simply opening the configuration assistant and clicking "Finish" at any point without having an "Original Prusa Mini" configured results in a crash for me.

If I add that printer, I'm able to make my changes without a crash (to include removing that printer). However, subsequent edits via the Configuration Assistant will crash.

bubnikv, 2022-01-05:
Would you please provide us with the content of our PrusaSlicer configuration directory, so we can open PrusaSlicer, open configuration wizard, follow a prescribed set of steps and get the crash?

We tested on two OSX laptops, one with Intel, the other M1 based with no crash.

bubnikv, 2022-01-07:
another one
https://github.com/prusa3d/PrusaSlicer/issues/7583#issuecomment-1001010658

bubnikv, 2022-01-07:
Would you please provide us with the content of our PrusaSlicer configuration directory, so we can open PrusaSlicer, open configuration wizard, follow a prescribed set of steps and get the crash?

We tested on two OSX laptops, one with Intel, the other M1 based with no crash. As of now we are stumbling in the darkness.

bubnikv, 2022-01-07:
Would you please try this debug build

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0+29-issue-7583-universal-gecd10dbc5-202201071631.dmg

You may need to enter

user prusaslicer
password slicer

if a login window opens

Follow these steps 

[image]
[image]

to open PrusaSlicer with command line console. After you open the wizard, you should get something similar to 

[image]

Just ignore the "[error]" comment, I just made these debugging outputs error outputs to get them out by default. If PrusaSlicer crashes, less than 27 of these lines will be printed out and we will know roughly where it crashes, so we will not stumble in darkness anymore.

Thank you.

ziwcam, 2022-01-07:
Results from Terminal.

Also, how can I send you my config? I've sanitized my API key, IP address, and recent files. Is there any other personal info in the configs I should be sure to remove?

```
[2022-01-07 09:58:42.966504] [0x000000010b754e00] [trace]   Initializing StaticPrintConfigs
[2022-01-07 09:58:43.496053] [0x000000010b754e00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[2022-01-07 09:58:52.083846] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 1
[2022-01-07 09:58:52.083867] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 2
[2022-01-07 09:58:52.083872] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 3
[2022-01-07 09:58:52.083894] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 4
[2022-01-07 09:58:52.083897] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 5
[2022-01-07 09:58:52.083901] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 6
[2022-01-07 09:58:52.083904] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 7
[2022-01-07 09:58:52.083907] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 8
[2022-01-07 09:58:52.083909] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 9
[2022-01-07 09:58:52.083912] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 10
[2022-01-07 09:58:52.083914] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 11
[2022-01-07 09:58:52.083917] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 12
[2022-01-07 09:58:52.083932] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 13
[2022-01-07 09:58:52.083936] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 14
[2022-01-07 09:58:52.083940] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 15
[2022-01-07 09:58:52.083943] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 16
[2022-01-07 09:58:52.083945] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 17
[2022-01-07 09:58:52.083950] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 18
[2022-01-07 09:58:52.083952] [0x000000010b754e00] [error]   ConfigWizard::priv::apply_config 19
zsh: segmentation fault  /Volumes/PrusaSlicer/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Saving session...

```

bubnikv, 2022-01-07:
@ziwcam thanks a lot, we may get there together. I will prepare another debug build tracing what happens after 19.

bubnikv, 2022-01-07:
@ziwcam Please try this one. Thanks.

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0+30-issue-7583-universal-g5010f0c2a-202201071726.dmg

ziwcam, 2022-01-07:
Unfortunately the message didn't change. Ran through twice to make sure I'd grabbed the right build, screenshot to show that the build was correct before this crash.

You can grab my config from here.

https://www.dropbox.com/s/xb0zs81hyfad9vx/CrashAndLog.zip?dl=0

```
[2022-01-07 11:07:38.059929] [0x000000011d035e00] [trace]   Initializing StaticPrintConfigs
[2022-01-07 11:07:38.556276] [0x000000011d035e00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[2022-01-07 11:08:12.308165] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 1
[2022-01-07 11:08:12.308202] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 2
[2022-01-07 11:08:12.308220] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 3
[2022-01-07 11:08:12.308226] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 4
[2022-01-07 11:08:12.308229] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 5
[2022-01-07 11:08:12.308235] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 6
[2022-01-07 11:08:12.308238] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 7
[2022-01-07 11:08:12.308241] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 8
[2022-01-07 11:08:12.308244] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 9
[2022-01-07 11:08:12.308246] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 10
[2022-01-07 11:08:12.308249] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 11
[2022-01-07 11:08:12.308261] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 12
[2022-01-07 11:08:12.308265] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 13
[2022-01-07 11:08:12.308268] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 14
[2022-01-07 11:08:12.308272] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 15
[2022-01-07 11:08:12.308275] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 16
[2022-01-07 11:08:12.308277] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 17
[2022-01-07 11:08:12.308292] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 18
[2022-01-07 11:08:12.308296] [0x000000011d035e00] [error]   ConfigWizard::priv::apply_config 19
zsh: segmentation fault  /Volumes/PrusaSlicer/Prus
<img width="2367" alt="Screen Shot 2022-01-07 at 11 08 05" src="https://user-images.githubusercontent.com/4970318/148587685-4a5347a3-d5b7-423a-a602-0451c3c2eab7.png">
aSlicer.app/Contents/MacOS/PrusaSlicer
Saving session...

```

ziwcam, 2022-01-07:
[image]

bubnikv, 2022-01-07:
@ziwcam 
Thanks. Having not received any further output is a hint as well. I may have found the bug.

bubnikv, 2022-01-07:
Hopefully fixed with https://github.com/prusa3d/PrusaSlicer/commit/6ff39ef07b200ed5ae6d0fc5847bbb0502ff8dbf
I will provide a build for testing in 20 minutes.

bubnikv, 2022-01-07:
@ziwcam Please try this build. It will still produce the debugging output, however it should work now without a crash. Thanks.

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0+31-issue-7583-universal-g6ff39ef07-202201071958.dmg

epetit91, 2022-01-07:
I am having the same problem in 10.12.6 Sierra.
When I open the configuration assistant and directly hit finish, PrusaSlicer crashes.

[Crash.txt](https://github.com/prusa3d/PrusaSlicer/files/7831234/Crash.txt)

ziwcam, 2022-01-07:
@bubnikv That seems to have worked. Thank you.

andy9a9, 2022-01-08:
@bubnikv should this solve #7583 as well?

bubnikv, 2022-01-08:
> @bubnikv <https://github.com/bubnikv> should this solve #7583
<https://github.com/prusa3d/PrusaSlicer/issues/7583> as well?

I think this build already contains fix for #7583, but the fix was
different, mentioned in #7583.

andy9a9, 2022-01-09:
OK, I can confirm, that now after clicking on "Finish" it didn't crash.
```
/Volumes/PrusaSlicer/PrusaSlicer.app/Contents/MacOS$ ./PrusaSlicer
[2022-01-09 17:22:10.057497] [0x0000000104a24580] [trace]   Initializing StaticPrintConfigs
[2022-01-09 17:22:10.351403] [0x0000000104a24580] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file)
[2022-01-09 17:22:25.954027] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 1
[2022-01-09 17:22:25.954055] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 2
[2022-01-09 17:22:25.954059] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 3
[2022-01-09 17:22:25.954062] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 4
[2022-01-09 17:22:25.954064] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 5
[2022-01-09 17:22:25.954067] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 6
[2022-01-09 17:22:25.954070] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 7
[2022-01-09 17:22:25.954071] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 8
[2022-01-09 17:22:25.954073] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 9
[2022-01-09 17:22:25.954074] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 10
[2022-01-09 17:22:25.954075] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 11
[2022-01-09 17:22:25.954083] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 12
[2022-01-09 17:22:25.954085] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 13
[2022-01-09 17:22:25.954087] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 14
[2022-01-09 17:22:25.954090] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 15
[2022-01-09 17:22:25.954092] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 16
[2022-01-09 17:22:25.954093] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 17
[2022-01-09 17:22:25.954099] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 18
[2022-01-09 17:22:25.954100] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 19
[2022-01-09 17:22:25.954103] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 20
[2022-01-09 17:22:25.954106] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 21
[2022-01-09 17:22:25.954108] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 22
[2022-01-09 17:22:25.954111] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 23
[2022-01-09 17:22:25.954118] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 24
[2022-01-09 17:22:25.954120] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 25
[2022-01-09 17:22:25.954122] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 26
[2022-01-09 17:22:25.954126] [0x0000000104a24580] [error]   ConfigWizard::priv::apply_config 27
```

bubnikv, 2022-01-10:
Fixed with c5d1e2449c25bb9c9af8116b40c6b0f00d138f7b

Thanks for all involved, especially to @ziwcam for help with debugging.

closed 2022-01-10 by bubnikv