prusa3d/PrusaSlicer #7656: Post-Processing-Script no longer working

bytefix, 2021-12-31:
### Version
Version 2.4.0+win64
### Operating system type + version
Windows 10

### Behavior
I'm using a post processing script developed by myself. It removes the oldest files from SD card, when gcode is saved to it.
Since 2.4 PrusaSlicer does not directly safes the file on the sd card but in a temp directonary and adds as argument: `C:\\Users\\jonas\\AppData\\Local\\Temp\\.11820.gcode.pp` and not the sd directonary.
The script needs the information about the choosed directonary (when pressed on export button) to delete the oldest files but there is no way to get this information anymore.

Possible fixes:
- use file saving like in 2.3 before
- add another argument of final file directonary

bubnikv, 2021-12-31:
https://github.com/prusa3d/PrusaSlicer/issues/7619#issuecomment-1001664242

bytefix, 2021-12-31:
Did not found that. Worked, thanks!

closed 2021-12-31 by bytefix