prusa3d/PrusaSlicer #7657: Raft expansion set to 0 still has expansion

MisterGlass, 2021-12-31:
### Version
Version 2.4.0+win64

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa Mini+

### Behavior
The file I am attempting to fit barely fits on my bed. When I turn on rafts and set expansion to 0, there is still some expansion and the print no longer fits on my bed

* _Expected Results_
A raft the same width as my model

* _Actual Results_
[image]

_Is this a new feature request?_ No.

#### Project File (.3MF) where problem occurs
[fabric.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7796789/fabric.3mf.zip)

n8bot, 2021-12-31:
There is also a first layer support expansion setting that is not set at zero in your 3MF (3 mm in your 3MF). Setting to zero solves the problem.

[image]

MisterGlass, 2021-12-31:
@n8bot  Thank you!

closed 2021-12-31 by MisterGlass