prusa3d/PrusaSlicer #7659: Problem when using the hollowing function for an SLA print

Blup1980, 2022-01-01:
### Version
PrusaSlicer
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 34,264MB
OpenGL installation
GL version:   4.6.0 NVIDIA 497.09
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
Not related to the bug: SL1

### Behavior
* When using the hollowing function on a part without support, the part seems to be drawn offset from its supposed location.
* The manually added holes seem to be placed not where the part is drawn but where the part is supposed to be drawn.

### Reproducing steps

1. Start PrusaSlicer (double click on the icon, not launched by file type opening)
2. Select the SL1 printer.
3. Untick "generate support" :
[image]

4. Ctrl-I to add a new part. Open the provided .3MF  DominoBody_resin_bug.3MF
5. Put the following options in the hollow and drill parameters:
[image]

6. The part is no longer on the buildplate:
[image]

7. Double click on the part to add a manual hole. The hole does not follow the part but seems to follow the position where the part is supposed to be:
[image]

8. Click "preview hollowed and drilled model". Two thing can occur depending where you put the hole:
     1. The following message appears:
      [image]
     2. The part is sliced correctly:
      [image]

#### Project File (.3MF) where problem occurs
[dominos.zip](https://github.com/prusa3d/PrusaSlicer/files/7797580/dominos.zip)

- PrusaSlicerFile.3mf is the Project file
- DominoBody_resin_bug.3MF is the file used to reproduce the bug

stepikovo, 2022-03-13:
Hi, I was able to reproduce it even on the newest 2.4.1 version, we will check it with the devs, thanks for reporting.

lukasmatena, 2022-03-14:
Thanks. This is in fact the same issue as #6100 (and others referenced from there). The fix is already on the way, it is just not tested yet.

closed 2022-03-14 by stepikovo