prusa3d/PrusaSlicer #7662: Preview Window Error after Slicing

brauschenberger, 2022-01-01:
Version: 2.4.0+linux-x64
Build: PrusaSlicer-2.4.0+linux-x64-202112211614
Operating System: Unix
System Architecture: 64 bit
System Version: Linux

5.4151-16008-gff376e5d5eal x86 64

Resident memory: 163MB
Shared memory: 82MB
Private memory: 81MB
Peak memory usage: 519MB
RAM size reserved for the Undo/Redo stack: 1,074MB
RAM size occupied by the Undo/Redo stack: 0MB

OpenGL, installation GL version: 3.1Mesa 20.3.5

3d Printer: Prusa SL1S

### Behavior
In Preview Window GUI throws an error which makes it nearly impossible to see 3d model. (see image attached)
[image]
[image]
[image]

lukasmatena, 2023-03-27:
Is it still happening in 2.6.0-alpha5? Thanks.

lukasmatena, 2023-07-10:
No response, quite likely fixed with the 2.6.0 changes. Closing.

closed 2023-07-10 by lukasmatena