prusa3d/PrusaSlicer #7663: prusaslicer v2.4.0 don't  run on  windows 7

JCD95-lab, 2022-01-01:
### Version
_Version of PrusaSlicer used goes here_ V.2.3.3

_Use `About->About PrusaSlicer` for release versions_V.2.4.0

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Windows 7 édition familiale premium
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  * After the installation Prusasclicer, i start it, but it stop immediately after the display of the first window.
  * after several trying of installation without result,  i must reinstal Prusasclicer v2.3.3 to have correct fonctionnality, but if i use the update i fall back on the problem.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2022-01-03:
Are you having problems with PrusaSlicer 2.3.3 or 2.4.0?
Does PrusaSlicer 2.4.0 run on your Windows 7 computer?
How does it "not run"?

JCD95-lab, 2022-01-03:
Prusaslicer v2.3.3 run correctly, only v2.4.0 stop at the beggining

bubnikv, 2022-01-03:
Isn't your issue the same as https://github.com/prusa3d/PrusaSlicer/issues/7574?

Does PrusaSlicer 2.4.0-beta1 run for you, but PrusaSlicer 2.4.0-beta2 does not?
Would you please test the builds at https://github.com/prusa3d/PrusaSlicer/issues/7574#issuecomment-1004078310 ?

JCD95-lab, 2022-01-03:
> Would you please test the builds at #7574 (comment)

I have to try....

JCD95-lab, 2022-01-03:
exact,
Prusaslicer 2.4.0.beta1  run

Prusaslicer 2.4.0.beta2 give window error

[image: image.png]

dartrax, 2022-01-03:
Your image wasn‘t uploaded correctly.

closed 2026-09-29 by prusaslicer-bot[bot]