prusa3d/PrusaSlicer #7664: Ramming porinting in the air

daviddam71, 2022-01-02:
### Version
2.3.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S & MMU2S (unmodified)

### Behavior
I recognize that printing in the air above the wipe tower is "standard" behavior. However, it causes a string of filament to stick to the hotend and causes problems on the next print. Is it possible to get the "ramming" done *on* the wipe tower instead at the highest level of the print when "No sparse layers" is selected for the wipe tower? 

Since this "feature" is considered standard behavior this would be a feature request.

bubnikv, 2022-01-03:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)#why-there-is-an-extrusion-hanging-in-the-air-over-a-multi-material-wipe-tower

closed 2022-01-03 by bubnikv

daviddam71, 2022-01-04:
As stated in my original issue, I know this is designed to help form the filament better for retraction. However, if the "No sparse layers" feature is turned on, couldn't it do the ramming *ON* the wipe tower instead of in the air? Also, as I stated originally, this behavior causes my next print to mess up since the string of extruded filament sticks to the hot end. While, it can be cleaned off by heating the hotend up, this shouldn't be necessary for such a sophisticated piece of software and hardware.