prusa3d/PrusaSlicer #7666: Preset Comparing Screen Edit and Save Feature Addition Request

sarvenn, 2022-01-02:
Compare Preset feature is quite usefull and userfriendly, however it is not complete unless we allow users to edit the values within the pop up screen and save the new values in that page. Otherwise users need to memories what to change in which line and then close the screen find the value to change in the preset page, etc.

My request is to add a save button above both presets that are being compared and the line value should be editable when double clicked.

[image]

bubnikv, 2022-01-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5384

sarvenn, 2022-01-03:
@bubnikv you closed my feature request as duplicate to an old request which is also closed. That request is implemented in 2.4 but what I request in this topic is to add the ability to change the values within that pop up screen which would be quite user friendly and easy to change the values between presets.
So, is my request including in a PR or created a backlog item?

bubnikv, 2022-01-03:
Sorry, I did not see the old issue was closed. 
I bet there is an open issue for your proposal, but to search each duplicate takes a significant time.

sarvenn, 2022-01-03:
#5384  is already open @bubnikv , I just wanted to know if my feature request is on the backlog list or is there any PR about this request. BTW, I would be happy if you could check my bug report #5734 🙏🏻

rayanth, 2022-08-10:
I'd also like to request a "transfer" button on EACH value, to transfer the value from one to the other.

closed 2026-09-29 by prusaslicer-bot[bot]