prusa3d/PrusaSlicer #7668: Incorrect first layer g-code being created in 2.4.0 when objects have very small sections

lps-rocks, 2022-01-02:
### Version
Version 2.4.0+win64

### Operating system type + version
PrusaSlicer
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 68,640MB
OpenGL installation
GL version:   4.6.0 NVIDIA 472.47
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3090/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 345 lines cut ...]
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### 3D printer brand / version + firmware version (if known)
3.10 - Prusa i3 mk3s+ w/BondTech extruder body + Mosquito Magnum Hotend running firmware from BondTech for hotend.
### Behavior
Incorrect first layer gcode is being produced. I've attached the model and settings export. This is new behavior in 2.4.0 - Version 2.3.3 slices the layer properly. 
* _Steps needed to reproduce the problem_
  * Load attached 3MF
* _Expected Results_
  * First layer includes entire model layer
* _Actual Results_
  * Only the first corner of the model is produced.
  * [image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[small container endcaps.zip](https://github.com/prusa3d/PrusaSlicer/files/7799164/small.container.endcaps.zip)

neophyl, 2022-01-02:
Turn your elephants foot compensation off.  That's bringing in all the first layer perimeters by 0.2mm.  After that happens the only perimeters big enough to print are the 2 very small islands.

lps-rocks, 2022-01-02:
It shouldn't be any taller than 2mm anywhere, it was extruded along the Z axis from a polyline drawn in 2D on the X/Y axis at a 0 z-axis height in AutoCAD. If it is, then there's a floating point error somewhere either in AutoCAD exporting the model to an STL or PrusaSlicer.

lps-rocks, 2022-01-02:
There's definitely some floating point error going on here. If I flip the model over, cut in PrusaSlicer at 2mm, I go from 'No Errors' to 873 open edges. Then if I repair the model, and flip it back over. It slices correctly.

bubnikv, 2022-01-04:
Your model has too thin walls to be printable. This is exacerbated by the elephant foot compensation at the 1st layer. If you disable elephant foot compensation, your model will slice the same way independent of whether you turn it upside down or not.

closed 2022-01-04 by bubnikv

lps-rocks, 2022-01-04:
Then why does leaving elephant foot enabled, flipping it over, cutting at 2mm then flipping it back fix the problem? The model measures 2mm height in my cad program and other slicers. 

This isn’t an elephant foot problem. This is a problem with Z-height floating point error causing problems in the elephant foot compensation routines.

bubnikv, 2022-01-04:
Maybe I don't understand what the problem is supposed to be. I see missing extrusions at the 1st layer and those are certainly because your model has too thin walls. Maybe more pictures with annotations will help.