prusa3d/PrusaSlicer #7671: Support Z Contact Distance

LuhBug23, 2022-01-02:
### Version
Version 2.3.3+x64

### Operating system type + version
2011 iMac 
High Seirra 10.13.6 (17G14042)

### 3D printer brand / version + firmware version (if known)
Ender 3 V2

### Behavior
* Changing Support Z contact distance without first removing "(detachable)" causes program to crash. 
* Set Z contact distance to "0.2mm (detachable)", change "0.2mm" to "0.175mm" keeping "(detachable)", the new text in Z contact distance drop down is "0.175 (detachable)". Hit enter to confirm the entry and the program will vanish before your eyes. Complete crash doesn't seem to otherwise lag or stress my computers resources to any degree. It is a very clean quick crash.

_Is this a new feature request?_
NO

rtyr, 2022-01-03:
Does it happen with current PrusaSlicer 2.4.0?

LuhBug23, 2022-01-03:
Just updated, I had not seen the new release. 

I attempted to recreate the error after updating and I received a pop-up that says "PrusaSlicer encountered an error". After clicking OK the Contact Z distance perimeter updates to the new value, .175 and the program proceeds as it should. No crash or error. 

Sorry and thank you for your help!

closed 2022-01-03 by LuhBug23