prusa3d/PrusaSlicer #7672: Error uploading to Prusa

jasonehines, 2022-01-03:
### Version
Version 2.4.0+arm64

### Operating system type + version
MacOS 12.1 (21C52)

### 3D printer brand / version + firmware version (if known)
Prusa MKS3+

### Behavior
I tried to upload a 3mf file to Prusa and get this error (not sure if it's a website issue or slicer):
Error: The uploaded file format is not supported or the file is corrupted
tempimageobgjgg.jpg
Please try to re-generate your gcode(s) with the newest PrusaSlicer version

#### Project File (.3MF) where problem occurs
[Frigidaire Stovetop bumper final.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7800124/Frigidaire.Stovetop.bumper.final.3mf.zip)

rtyr, 2022-01-03:
I was not able to reproduce it on win64.

jasonehines, 2022-01-11:
This appears to be an issue with Safari. I was abled to get it to upload via Edge.

closed 2022-01-11 by jasonehines