prusa3d/PrusaSlicer #7674: I have a problem g-codes generated from slicer, for ender 3 v2

BeardedEmiloss, 2022-01-03:
### Version[USB_2h24m_0,16mm_215C_PLA_ENDER3V2.zip](https://github.com/prusa3d/PrusaSlicer/files/7801725/USB_2h24m_0.16mm_215C_PLA_ENDER3V2.zip)_Version of PrusaSlicer used 2.4.0+win64_

### Operating system Windows 10 64-bit home

### 3D printer brand / Ender-3 v2 4.2.2mainboard（32bit）Marlin2.0.6.1V1.0.4HW_4.2.2_20210803

### Behavior
* _Describe the problem_
I have a problem g-codes generated from slicer, for ender 3 v2 started to break down. They started printing in the air and in strange places. I don't know what happened.
[USB_2h24m_0,16mm_215C_PLA_ENDER3V2.zip](https://github.com/prusa3d/PrusaSlicer/files/7801725/USB_2h24m_0.16mm_215C_PLA_ENDER3V2.zip)

rtyr, 2022-01-03:
The provided g-code seems OK. The behavior you are describing usually happens with corrupted g-code file. You may try different USB/SD card or at least format it.

rtyr, 2022-06-20:
No response. Closing.

closed 2022-06-20 by rtyr