prusa3d/PrusaSlicer #7675: Calculation relation between spacing and density of honeycomb filling

Ccaomei, 2022-01-03:
Hello everyone, I would like to know the calculation relationship between the spacing and density of honeycomb filling, and where to set the offset distance between the filling graph and polygon, I hope you can give me your advice, thank you very much.

n8bot, 2022-01-04:
https://github.com/prusa3d/PrusaSlicer/blob/a554f1fb089ccbc6fee995571091054703ca181d/src/libslic3r/Fill/FillHoneycomb.cpp#L16-L33

Ccaomei, 2022-01-04:
> https://github.com/prusa3d/PrusaSlicer/blob/a554f1fb089ccbc6fee995571091054703ca181d/src/libslic3r/Fill/FillHoneycomb.cpp#L16-L33

@n8bot 
hi，What are the units of polygon coordinates?Do you have documentation for information about units。thanks !

n8bot, 2022-01-04:
To be honest, I have no more information. I've never needed to change the honeycomb. I assume the coordinates are in mm, but they may be scaled mm, which would be x/0.000001, or a million times larger.

Ccaomei, 2022-01-05:
> To be honest, I have no more information. I've never needed to change the honeycomb. I assume the coordinates are in mm, but they may be scaled mm, which would be x/0.000001, or a million times larger.

@n8bot , hi , In the demo, I set spacing and density, but the density of the filling graph is not the specified density. Can you explain the relationship between spacing and density and the offset of the filling graph?

n8bot, 2022-01-05:
I'm no expert by any means, but I suspect you might be forgetting to use scaled units (they will never have a decimal place. Units are scaled to integer-like values. 1,000,000 larger than the "real" value.)

The distance, spacing, etc is taken from the `const FillParams &params,` that are sent as an argument to the _fill_surface_single().

https://github.com/prusa3d/PrusaSlicer/blob/a554f1fb089ccbc6fee995571091054703ca181d/src/libslic3r/Fill/FillBase.hpp#L33-L60

n8bot, 2022-01-05:
But, the spacing is derived from the Fill -- which is what should be calling this _fill_surface_single.

So, without knowing more details about what you're doing it's difficult to help you. It all changes depending on what and where you are trying to do it.

n8bot, 2022-01-05:
I suggest you forget about the function we've been talking about, and look how the support material uses the fillers. It seems much easier to adapt as needed:

https://github.com/prusa3d/PrusaSlicer/blob/934611206d8ab360ccabe9f27c7d6d72f3fb807e/src/libslic3r/SupportMaterial.cpp#L3234-L3254

https://github.com/prusa3d/PrusaSlicer/blob/481def3205182193140c6186420fe97adbda8abc/src/libslic3r/Fill/FillBase.cpp#L81-L95

Ccaomei, 2022-01-06:
[image]

@n8bot Hi, I want to refer to demo, write an interface, directly pass polygons, see the filling effect.
I'd like to ask some questions:
Fill_params. density = float(filler-&gt;Spacing / 50.0);What does spacing mean, and what does 50.0 mean?
2, is there any parameter to set the filling line width;
3. The relationship between filling density and filling line width;
Help to answer again, extremely grateful!!

n8bot, 2022-01-06:
1. spacing is the distance, center-to-center, between two lines, if they are printed right next to each other.
   - The 50 is just a value used to make a density of 0.1, based on the testing scenario 5 mm spacing.
1. The line width is basically the same as the spacing. 
1. The density is related to the spacing. The spacing relates closely to the line width.

The line width is really just an idea, though. The polyline coordinates are based on the centerlines of the extrusions. The height and width of the lines are basically metadata that is used in other places. Think about the gcode output that PS is doing: it defines the X, Y, and Z coordinates of the polylines explicitly. But the thickness in height and width is determined implicitly by the extrusion amounts. The width and height are essentially decoupled.

Ccaomei, 2022-01-06:
[image]
@n8bot ,hi,I set spacing=5mm and density=0.5. The SVG image generated at this density, with the padding line width set to 5mm, should have 1:1 padding line width and padding line spacing, but the resulting SVG image looks like this, what am I wrong?

n8bot, 2022-01-06:
I'm not sure what information you need.

The spacing + density define how far apart those lines are.

The width of the lines is not set or relevant to these tests. They are testing the infill pattern, not the extrusion width. No extrusion width is used in these tests. If you want the SVG output lines to be different, you will have to edit the SVG. If the SVG is using a stroke value, this will be easy. Simply increase the stroke to your liking.

But this is getting farther from my knowledge of PrusaSlicer, and into the realm of general knowledge.

Ccaomei, 2022-01-07:
> I'm not sure what information you need.
>
> The spacing + density define how far apart those lines are.
>
> The width of the lines is not set or relevant to these tests. They are testing the infill [...]

@n8bot ，hi，I want to express that the SVG diagram I got is not consistent with the density I set. I want to know what is wrong and I hope to get your help. The SVG diagram and code have been pasted, please help to have a look.Thank you very much!

n8bot, 2022-01-07:
The width of the SVG lines does not relate to anything, even if you multiply them by a million it seems.

I have sliced something in the software with 0.5 mm spacing, and 50% density, which should provide a similar result to yours. When I scale it and overlay it against your SVG, we can see what is happening and how the line thickness of the SVG is  misleading.

[image]

[image]

n8bot, 2022-01-07:
The black rectangle is defining only the inner part, there is no line width accounted for on its edge. The yellow you see in the overlaid image is the perimeters -- as you see, the black line of the rectangle in your SVG is on the edge of the inside of the perimeter.

Ccaomei, 2022-01-07:
@n8bot ，hi，“ filler->angle = float(-(PI)/2.0);  ” ，May I ask how to set the Angle so that I can set the filling Angle from 0 degrees to 360 degrees?

n8bot, 2022-01-07:
Yes. The angle is set in radians. So, if I remember correctly, a value of pi is 180 degrees.

Ccaomei, 2022-01-11:
@n8bot ,hi, I would like to ask you another question. Where is the parameter "fill/contour overlap" set in the fill module?

n8bot, 2022-01-11:
The overlap is done differently depending on the infill type. I like the way rectilinear does it, with an `ExPolygonWithOffset`

https://github.com/prusa3d/PrusaSlicer/blob/a6d280bd9ecc131ff0589a6246be1ecdea0fb0a7/src/libslic3r/Fill/FillRectilinear.cpp#L3020-L3042

It is essentially just an offset of the input polygon for the fill.

closed 2022-01-11 by lukasmatena

Ccaomei, 2022-01-11:
@n8bot, hi, I do not understand the use of "ExPolygonWithOffset", could you please give me a demo to use it to offset the filling line? If it is honeycomb filling, how to offset the filling line? Looking forward to your reply. thank you

n8bot, 2022-01-12:
For the honeycomb infill. the input ExPolygon itself is offet. The `ExPolygonWithOffset` is a special type that contains information about offset polygons, which seems made for the purpose of accounting for the overlap of infill.

See line 84: 
https://github.com/prusa3d/PrusaSlicer/blob/1dffc40f324a9bf2361b55c0f4e227508e25f388/src/libslic3r/Fill/FillBase.cpp#L81-L95