prusa3d/PrusaSlicer #7677: What Happens When Dimensions of Print Bed And Build Volume Do Not Match

rf1k-mjh11, 2022-01-03:
### Version

2.4.0 beta1 win64 (and **presumably earlier versions**)

### Operating system type + version

Windows 8.1

### 3D printer brand / version + firmware version (if known)

Printer: Conrad Renkforce RF1000, 
Firmware: RF.01.47 (development) - modified.  This is a customized, Repetier-based firmware specifically made for the printer.

### Behavior

Background information:
The RF1000, RF2000 printers were designed to have about 17-20mm of free space in Y, at the front of the bed.  So the bed surface actually starts at X=0, Y=20.  The reason for this unusual approach is unknown, presumably to make nozzle purging and cleaning, etc, easier.
The printer's actual travel values are X=245, Y=245 and Z=200, which allows a build volume as shown in this picture:
[image]

### The Problem:
Unfortunately, Prusa Slicer **will not slice anything** that protrudes in front of the print bed, even if it is within the printable volume.

OK, I would agree that Prusa Slicer should insure the first layer of a print does not protrude beyond the bed surface.  But, beyond that, limiting build volume to the projected bed surface is 'stealing' potential build volume from the user.

Prusa Slicer could, theoretically, determine the dimension 'W' shown in the picture, and use that as a limiting criteria.  But I think the User should bear a little responsibility, too.  Let them worry about 'W'.
But don't take away more than **_0.75 liters !!_** of build volume.

An possible **work-around** would be to show a warning and giving the user the option to slice anyway.

BTW: Slic3r, v1.2.9, doesn't have a problem with the issue at all.  The user could actually place the object **completely outside** of the print bed and Slic3r will still generate GCode.

#### Project File (.3MF) where problem occurs

N/A

bubnikv, 2022-01-03:
You can increase the size of the print bed and load a texture where you paint the smaller rectangle where you should print.

rf1k-mjh11, 2026-09-18:
I agree that you could use that approach with a texture.  I currently use a similar approach to avoid an area where the glass ceramic bed has a small, shallow chip missing.

It is a doable approach (your suggestion), but requires every user (that has a similar issue - of which there may be very few) to be knowledgeable enough to use the approach - whereas solving it via slicer would be more of a Plug & Play solution.

But it is OK if you disregard this condition.  As I said, there are probably very few people it affects.

Close the issue if you want.

Sorry to have bothered you,

rf1k-mjh11