prusa3d/PrusaSlicer #7678: Typo / wrong translation into German

21mac, 2022-01-03:
### Version
2.2.0

### Operating system type + version
Windows 10 Ultimate x64 

### 3D printer brand / version
Original Prusa i3 MK3S+ with Octoprint 1.7.2 (octoprint image on the prusa website, then updated)

### Behavior
After slicing I upload the gcode via "CTRL + SHIFT + G" (or the button in the right lower corner) and when there is an error, the following message is displayed:
[image]

There is a typo in "hochloden", it has to be "hochladen" (o --> a)

dartrax, 2022-01-03:
Correct. Also, it would be better to write „zum Druckhost“.

bubnikv, 2022-01-04:
Is the wording still incorrect in PrusaSlicer 2.4.0?

dartrax, 2022-01-04:
Yes:
[image]

fe60, 2022-01-17:
Using PoEdit [1] leads to a lot of additional changes [2] instead of changing only one entry [3]. Hence I don't trust the generated .mo-file and I don't create a pull request.

[1] https://github.com/prusa3d/PrusaSlicer/blob/master/doc/Localization_guide.md
[2] https://github.com/fe60/PrusaSlicer/commit/bd83e6da81d8c00e5efa0e993cd1f08255a4aac5#diff-506339c6b261abed22dbe8b733e7cc6d7abf0a827dac5abfbd73e34159d37b27
[3] https://github.com/fe60/PrusaSlicer/commit/bb240c81e390b4b8c81503cce73ab47f34412f21

21mac, 2022-01-17:
> Using PoEdit [1] leads to a lot of additional changes [2] instead of changing only one entry [3]. Hence I don't trust the generated .mo-file and I don't create a pull request.
> 
> [1] https://github.com/prusa3d/PrusaSlicer/blob/master/doc/Localization_guide.md [2] [fe60@bd83e6d#diff-506339c6b261abed22dbe8b733e7cc6d7abf0a827dac5abfbd73e34159d37b27](https://github.com/fe60/PrusaSlicer/commit/bd83e6da81d8c00e5efa0e993cd1f08255a4aac5#diff-506339c6b261abed22dbe8b733e7cc6d7abf0a827dac5abfbd73e34159d37b27) [3] [fe60@bb240c8](https://github.com/fe60/PrusaSlicer/commit/bb240c81e390b4b8c81503cce73ab47f34412f21)

After checking the differences in the changed file it seems to be correct now.
Thanks

closed 2022-01-17 by 21mac

fe60, 2022-01-17:
@21mac 
None of my commits reached the master branch of the original repository. So it's still wrong there ([PrusaSlicer_de.po#L3530](https://github.com/prusa3d/PrusaSlicer/blob/master/resources/localization/de/PrusaSlicer_de.po#L3530)).
I wanted a member of the dev team to decide how to integrate the proposed correction.

@bubnikv 
Can you please check my [comment above](https://github.com/prusa3d/PrusaSlicer/issues/7678#issuecomment-1014239633) and let us know, how to proceed?

lukasmatena, 2022-01-20:
@fe60 It should be ok after we updated the dictionaries (d1523d2a6b94f43ad55ecf802e0159b84ff15b3a). Thanks.