prusa3d/PrusaSlicer #7679: How to pass multiple sliced polygons at once when using Infill

Ccaomei, 2022-01-04:
Hi everyone, how to pass multiple sliced polygons at once when using Infill, because I see that the filling interface parameter is Exploygon, not Ploygons. I would appreciate your guidance.

n8bot, 2022-01-04:
Where are you trying to do infill and in which way? There are several different ways the infillers are called. Well, two at least: for infill, and for support material.

There are `ExPolygon`, `Polygon` and `ExPolygons` and "Polygons`. Sometimes, when you do an operation on one polygon, or expolygon, a collection is returned instead of a single.

Also, sometimes a particular function has no overload for a given scenario. I.e. it expects a collection of the particular type, not a single one.

If you provide specifics about what you're trying to do, and where in the code you're doing it, I can help you out.

Ccaomei, 2022-01-04:
> 

When a slice has multiple polygons, I want to pass multiple ploygons directly to the infill interface without doing any analysis on whether polygons are nested.What do I do at this point, because the parameter filling the interface is Exploygon

n8bot, 2022-01-04:
There are lots of different polygons and expolygons flying around in the source code. You'll have to be more specific.

Ccaomei, 2022-01-04:
> There are lots of different polygons and expolygons flying around in the source code. You'll have to be more specific.

@n8bot
hi,  If I have a slice, two polygons, and a polygon inside the two polygons, and I need to fill the slice, then when I call the fill interface, I want to do nothing and pass in the four polygons ploygons directly, but the actual filling interface looks like this: 
  "void _fill_surface_single(  const FillParams  &params, unsigned int    thickness_layers,  const std::pair<float, Point>   &direction, ExPolygon  expolygon, Polylines      &polylines_out) ;"  ，The interface is Expolygon. In this case, what should I do?

n8bot, 2022-01-04:
You will have to play some tricks. Really, once an ExPolygon is turned into a Polygon, it's not meant to go back to an ExPolygon.

You can easily turn a collection of Polygons into a collection of ExPolygons. But there is no such function to convert a single polygon, as far as I can tell.

What you can do is make a collection of polygons from your single polygons, and then use the function `ExPolygons to_expolygons(const Polygons &polys)` to make a collection of ExPolygons from that collection of Polygons. From there, you can iterate through the collection of ExPolygons, passing them off to the infiller like you wish:

```cpp
    Slic3r::ExPolygons expp = to_expolygons(your_newly_created_collection_of_polygons);
    // Create the infills for each of the regions.
    Polylines polylines_out;
    for (size_t i = 0; i < expp.size(); ++ i)
        _fill_surface_single(
            params,
            surface->thickness_layers,
            _infill_direction(surface),
            std::move(expp[i]),
            polylines_out);;
```

n8bot, 2022-01-04:
FYI, to make the collection of polygons, you just make a Polygons type, which is a  vector, of the single polygons. Same as you'd make any std::vector but use the Polygons type.

Ccaomei, 2022-01-04:
> You will have to play some tricks. Really, once an ExPolygon is turned into a Polygon, it's not meant to go back to an ExPolygon.
> 
> You can easily turn a collection of Polygons into a collection of ExPolygons. But there is no such function to convert a single polygon, as far as I can tell.
> 
> What you can do is make a collection of polygons from your single polygons, and then use the function `ExPolygons to_expolygons(const Polygons &polys)` to make a collection of ExPolygons from that collection of Polygons. From there, you can iterate through the collection of ExPolygons, passing them off to the infiller like you wish:
> 
> ```c++
>     Slic3r::ExPolygons expp = to_expolygons(your_newly_created_collection_of_polygons);
>     // Create the infills for each of the regions.
>     Polylines polylines_out;
>     for (size_t i = 0; i < expp.size(); ++ i)
>         _fill_surface_single(
>             params,
>             surface->thickness_layers,
>             _infill_direction(surface),
>             std::move(expp[i]),
>             polylines_out);;
> ```
@n8bot ，
hi，Where is "ExPolygons to_expolygons(const Polygons &polys)", I did not find it, please help tell me the address;Another question: is to_expolygons a vector of all polygons in a slice?

Ccaomei, 2022-01-04:
[image]

When a slice has multiple polygons (some of which are nested), I don't do anything about it, save it directly to Ploygons, then call the to_exPolygons interface and iterate through the for loop, which is not what I want. What I need is: when the slice has more than one nesting, the filling result is that the inner polygon is not filled. I pass in the six polygons ploygons as I drew them, and what I get is the fill line.

n8bot, 2022-01-04:
You will have to subtract the inner polygons from the outer polygon before you send them to the infiller. 

Use diff() to do that. I suppose you will have to do it multiple times, for each inner polygon. 

So, I suggest making collection of polygons for the inner polygons that you will subtract. Then iterate through that collection, using the diff command to subtract them one by one from the main polygon.

ledvinap, 2022-01-04:
@Ccaomei:  ExPolygon is outer `Polygon` + multiple hole `Polygons`). You can iterate over `ExPolygons` without loosing information (IIRC ExPolygons must not intersect).

Clipper library is capable of normalizing polygons. 
`Slic3r::ExPolygons union_ex(const Slic3r::Polygons &subject, ClipperLib::PolyFillType fill_type = ClipperLib::pftNonZero);` may be good starting point, it will union outer polygons on overlap and handle holes correctly.

n8bot, 2022-01-04:
Nice! I never thought of using union_ex... I didn't realize it could accept polygons. I never thought to look!

But, the problem I think the OP will have is that he is not starting with genuine ExPolygons, so he will have to create the expolygons himself and define the holes manually.

Ccaomei, 2022-01-05:
> @Ccaomei: ExPolygon is outer `Polygon` + multiple hole `Polygons`). You can iterate over `ExPolygons` without loosing information (IIRC ExPolygons must not intersect).
>
> Clipper library is capable of [...]

@n8bot @ledvinap  hi，In the case I listed, the six Polygons were deposited into one polygons,Then call "Slic3r::ExPolygons union_ex" (const Slic3r::Polygons &subject,ClipperLib::PolyFillType fill_type = ClipperLib::pftNonZero);"Method, you can get Expolygons.Then iterate through each Expolygon with a for loop, right?

n8bot, 2022-01-05:
It seems like that should work, yes. I have not gone "deep enough" into the clipper libraries to know that the union_ex() identifies the holes. I also can not tell for sure if void FillHoneycomb::_fill_surface_single() will properly avoid the holes of the expolygon. However, intuitively, assuming that everything works as we think it does, it should work exactly as you want. Yes.

ledvinap, 2022-01-05:
Only problem may be orientation of hole polygons with respect to contour - `pftNonZero` will work only if hole orientation is opposite. Documentation is here: http://www.angusj.com/delphi/clipper/documentation/Docs/Units/ClipperLib/Types/PolyFillType.htm

Ccaomei, 2022-01-06:
@ledvinap hi，I'd like to ask some questions:
1, Fill_params. density = float(filler->Spacing / 50.0);What does spacing mean, and what does 50.0 mean?
2, is there any parameter to set the filling line width;
3. The relationship between filling density and filling line width;
Help to answer again, extremely grateful!
Do you have relevant documents for setting filling parameters for my reference.

closed 2026-09-29 by prusaslicer-bot[bot]