prusa3d/PrusaSlicer #7680: [Feature request] Inverse Motion option per extruder

woogerboy21, 2022-01-04:
### Version
2.4.0+

For those of us using both sides of an extruder gear to push/pull filament it would be nice to have a checkbox option under the extruder settings to inverse the extruder value.  For example if extruder 1 uses the right side of the gear with the output facing upwards it would rotate properly but extruder 2 uses the other side of gear on the same motor shaft it needs to rotate the opposite direction to push filament.  With this option prusaslicer would generates the gcode for extruder 1 by writing E-100 to move filament forward instead of writing E100 with possibly a comment in the gcode mentioning inverted motion for clarity.

The other configuration in which this option would be handy is if your extruder is a right handle extending extruder rather than the commonly used left extending handle (which also uses the left side of the gear).

n8bot, 2022-01-04:
What firmware are you using? I know that RRF can easily implement this into their tool definitions with simple G-Code. I'm sure marlin can also easily be made to achieve this.

This screams like something a firmware should handle, not the slicer.

bubnikv, 2022-01-04:
> This screams like something a firmware should handle, not the slicer.

Exactly.

woogerboy21, 2022-01-04:
The printers I use have marlin on them.  And I would have to disagree with the firmware approach. Many users have no interest in creating custom firmware.  And yes you could go about writing something like a python script to reverse the gcode that is generated but all of these items are additions to something that would be much simpler for the end user to just check a box in a slicer.  The goal with this request was to add additional functionality that would be easy for users to get results.

closed 2026-09-29 by prusaslicer-bot[bot]